al177/esp8089 #3: Current driver status, possible upstreaming?

noahwilliamsson, 2016-07-29:
There seem to be attempts to clean up and upstream the driver in forks of this repo.  I've found:

- @jwrdegoede's tree at https://github.com/jwrdegoede/esp8089
- @Icenowy's cleanup branch at https://github.com/Icenowy/esp8089/tree/cleanup with corresponding LKML posts at:
    - [[PATCH] staging: esp8089: add new driver](https://marc.info/?t=146633172000001&r=1&w=3)
    - [[PATCH v2] staging: esp8089: add new driver](https://marc.info/?t=146634028700001&r=1&w=2)

Since issue reporting is off by default on GitHub forks I can't leave a comments on the specific repos.  Instead I've opted to summarize things here and ping those involved.  Personally I'd be happy to see the driver being upstreamed as it's a cheap (and low power) way of getting WiFi connectivity on Raspberry Pi Zeroes.

(sorry for the wide table below..)

| Tree / Status                            | https://github.com/al177/esp8089         | https://github.com/jwrdegoede/esp8089    | https://github.com/Icenowy/esp8089/tree/cleanup |
| ---------------------------------------- | ---------------------------------------- | ---------------------------------------- | ----------------------------------------------- |
| Last updated                             | 2016-05-26                               | 2016-07-10                               | 2016-06-19                                      |
| Linux v4.6 `.ampdu_action` fix              | NO                                       | jwrdegoede/esp8089@865e748f060f4bcbefc0c16aa786e37551703bf3 | NO, frame aggregation support dropped(!)        |
| Linux v4.7 `NL80211_BAND_2GHZ` support     | NO                                       | jwrdegoede/esp8089@74af6ac9d2c8b1c6d03d34b74136e29b7fd5b977 | 364ee3b64344c926636e35fd84dc51a44039ba3b        |
| Support for non-bleeding edge kernels    | YES, LINUX_VERSION_CODE guards           | YES, LINUX_VERSION_CODE guards           | NO, dropped during upstreaming attempt          |
| DKMS support                             | al177/esp8089@41ed6ea97ab90f9a323e2bb34f87b92daa26b3f6 | jwrdegoede/esp8089@41ed6ea97ab90f9a323e2bb34f87b92daa26b3f6 | NO, but `make -C $KDIR M=$PWD` supposed to work |
| Bugfix: `<../net/mac80211/ieee80211_i.h>`  | al177/esp8089@097a1ef0683d09dc29eb57a22a53fb49794a5fab | jwrdegoede/esp8089@097a1ef0683d09dc29eb57a22a53fb49794a5fab | NO                                              |
| Bugfix: transposed memset params         | NO                                       | NO                                       | Icenowy/esp8089@79351b14abaca2915ab2729c758ce337616cfcf8        |
| Firmware loading via `request_firmware()`  | NO                                       | jwrdegoede/esp8089@6b100f25836fcdd927bc07452c625189eab2a2b9 | 1cd2da85c1cb32252ee150f77bd96573d8ea980c        |
| Firmware built-in (.h) support           | YES                                      | YES                                      | NO, dropped during upstreaming attempt          |
| Firmware in eagle_fw{1,2,3}.h            | YES                                      | YES                                      | YES (should really be dropped)                  |
| Firmware .bin files in repo              | NO                                       | jwrdegoede/esp8089@4445df9c63be5d92268d0304b2adfeb723c9f87f | Icenowy/esp8089@711334e24668d4660e43d4ad2e3cf8c054b7a551        |
| `MODULE_FIRMWARE()` documentation          | NO                                       | NO                                       | NO                                              |
| Dropped support for not `P2P_CONCURRENT` | NO                                       | NO                                       | Icenowy/esp8089@711334e24668d4660e43d4ad2e3cf8c054b7a551        |
| Dropped support for `testmode.c`         | NO                                       | jwrdegoede/esp8089@dd28cb55a1b67c3fc81ef63617eee9c44b6f21ec | Icenowy/esp8089@20f0ff05bf1431f8fc23d0f3bf6b86b91cef82c2        |
| Dropped support for `FAST_TX_STATUS`     | NO                                       | NO                                       | Icenowy/esp8089@8cd5735bedee11fb1861a880fb16f9693816594a        |
| Dropped support for Android, SPI interface  | NO                                       | NO                                       | Icenowy/esp8089@909c1f673f16684ed32d2194f225367cedfc510d, Icenowy/esp8089@93f574bc689cd125c9f15c93593311bb0d19e91a |
| `esp_reset_gpio` modules parameter       | al177/esp8089@17f726ab5e14063db0996aabef22cfcf3a2ef653, al177/esp8089@c01f5bde78c6bc15ff4cf5e2100805be2c76ba2b, al177/esp8089@879a31f0e7da0f6d450f171b74ae43f1a2c3f684 | Dropped in jwrdegoede/esp8089@a9fd9d6ab40f4c74a3ed6e55901f642e2fd299d1 because Device tree | NO, support never merged |

@Icenowy: The merge window for 4.8 is currently open.  Will you submit a V3 patch?  If yes, please consider:

- Reverting the drop of frame aggregation in Icenowy/esp8089@79351b14abaca2915ab2729c758ce337616cfcf8 and apply the fix from jwrdegoede/esp8089@865e748f060f4bcbefc0c16aa786e37551703bf3
- Applying the fix of `#include <../net/mac80211/ieee80211_i.h>` in `esp_mac80211.c` (see al177/esp8089@097a1ef0683d09dc29eb57a22a53fb49794a5fab)
- Add `MODULE_FIRMWARE()` macros so the required firmwares are listed properly by modinfo
- Drop `eagle_fw1.h`, `eagle_fw2.h` and `eagle_fw3.h` since they are no longer needed with external firmware files
- (should `esp_init_data.h` be converted into a firmware file with a corresponding request_firmware() too? I think it's just settings and not machine code though)
- Drop the commented out `esp_readwrite_file()` call
- Drop code guarded by `#ifdef TEST_MODE`

Icenowy, 2016-07-29:
Thanks. I will consider it this evening (UTC+8)

Tree / Statushttps://github.com/al177/esp8089https://github.com/jwrdegoede/esp8089https://github.com/Icenowy/esp8089/tree/cleanupLast updated2016-05-262016-07-102016-06-19Linux v4.6 .ampdu_action fixNOjwrdegoede/esp8089@865e748NO, frame aggregation support dropped(!)Linux v4.7 NL80211_BAND_2GHZ supportNOjwrdegoede/esp8089@74af6ac364ee3bSupport for non-bleeding edge kernelsYES, LINUX_VERSION_CODE guardsYES, LINUX_VERSION_CODE guardsNO, dropped during upstreaming attemptDKMS supportal177/esp8089@41ed6eajwrdegoede/esp8089@41ed6eaNO, but make -C $KDIR M=$PWD supposed to workBugfix: <../net/mac80211/ieee80211_i.h>
al177/esp8089@097a1efjwrdegoede/esp8089@097a1efNOBugfix: transposed memset paramsNONOIcenowy/esp8089@79351b1Firmware loading via request_firmware()
NOjwrdegoede/esp8089@6b100f21cd2da8Firmware built-in (.h) supportYESYESNO, dropped during upstreaming attemptFirmware in eagle_fw{1,2,3}.hYESYESYES (should really be dropped)Firmware .bin files in repoNOjwrdegoede/esp8089@4445df9Icenowy/esp8089@711334e
NONOIcenowy/esp8089@711334eDropped support for testmode.c
NOjwrdegoede/esp8089@dd28cb5Icenowy/esp8089@20f0ff0Dropped support for FAST_TX_STATUS
NONOIcenowy/esp8089@8cd5735Dropped support for Android, SPI interfaceNONO
Icenowy/esp8089@909c1f6, Icenowy/esp8089@93f574b

al177/esp8089@17f726a, al177/esp8089@c01f5bd, al177/esp8089@879a31f
Dropped in jwrdegoede/esp8089@a9fd9d6 because Device treeNO, support never merged

Reverting the drop of frame aggregation in Icenowy/esp8089@79351b1 and apply the fix from jwrdegoede/esp8089@865e748
Applying the fix of #include <../net/mac80211/ieee80211_i.h> in esp_mac80211.c (see al177/esp8089@097a1ef)Add MODULE_FIRMWARE() macros so the required firmwares are listed properly by modinfoDrop eagle_fw1.h, eagle_fw2.h and eagle_fw3.h since they are no longer needed with external firmware files(should esp_init_data.h be converted into a firmware file with a corresponding request_firmware() too? I think it's just settings and not machine code though)Drop the commented out esp_readwrite_file() callDrop code guarded by #ifdef TEST_MODE

jwrdegoede, 2016-07-29:
Hi,

On 29-07-16 07:44, noahwilliamsson wrote:
> There seem to be attempts to clean up and upstream the driver in forks of this repo. I've found:
>
>   * @jwrdegoede <https://github.com/jwrdegoede>'s tree at https://github.com/jwrdegoede/esp8089
>   * @Icenowy <https://github.com/Icenowy>'s cleanup branch at https://github.com/Icenowy/esp8089/tree/cleanup with corresponding LKML posts at:
>       o [PATCH] staging: esp8089: add new driver <https://marc.info/?t=146633172000001&r=1&w=3>
>       o [PATCH v2] staging: esp8089: add new driver <https://marc.info/?t=146634028700001&r=1&w=2>
>
> Since issue reporting is off by default on GitHub forks I can't leave a comments on the specific repos. Instead I've opted to summarize things here and ping those involved. Personally I'd be happy to see the driver being upstreamed as it's a cheap (and low power) way of getting WiFi connectivity on Raspberry Pi Zeroes.
>
> (sorry for the wide table below..)
>
> Tree / Status 	https://github.com/al177/esp8089 	https://github.com/jwrdegoede/esp8089 	https://github.com/Icenowy/esp8089/tree/cleanup
> Last updated 	2016-05-26 	2016-07-10 	2016-06-19
> Linux v4.6 |.ampdu_action| fix 	NO 	jwrdegoede/esp8089@865e748 <https://github.com/jwrdegoede/esp8089/commit/865e748f060f4bcbefc0c16aa786e37551703bf3> 	NO, frame aggregation support dropped(!)
> Linux v4.7 |NL80211_BAND_2GHZ| support 	NO 	jwrdegoede/esp8089@74af6ac <https://github.com/jwrdegoede/esp8089/commit/74af6ac9d2c8b1c6d03d34b74136e29b7fd5b977> 	364ee3b <https://github.com/al177/esp8089/commit/364ee3b64344c926636e35fd84dc51a44039ba3b>
> Support for non-bleeding edge kernels 	YES, LINUX_VERSION_CODE guards 	YES, LINUX_VERSION_CODE guards 	NO, dropped during upstreaming attempt
> DKMS support 	al177/esp8089@41ed6ea <https://github.com/al177/esp8089/commit/41ed6ea97ab90f9a323e2bb34f87b92daa26b3f6> 	jwrdegoede/esp8089@41ed6ea <https://github.com/jwrdegoede/esp8089/commit/41ed6ea97ab90f9a323e2bb34f87b92daa26b3f6> 	NO, but |make -C $KDIR M=$PWD| supposed to work
> Bugfix: |<../net/mac80211/ieee80211_i.h>| 	al177/esp8089@097a1ef <https://github.com/al177/esp8089/commit/097a1ef0683d09dc29eb57a22a53fb49794a5fab> 	jwrdegoede/esp8089@097a1ef <https://github.com/jwrdegoede/esp8089/commit/097a1ef0683d09dc29eb57a22a53fb49794a5fab> 	NO
> Bugfix: transposed memset params 	NO 	NO 	Icenowy/esp8089@79351b1 <https://github.com/Icenowy/esp8089/commit/79351b14abaca2915ab2729c758ce337616cfcf8>
> Firmware loading via |request_firmware()| 	NO 	jwrdegoede/esp8089@6b100f2 <https://github.com/jwrdegoede/esp8089/commit/6b100f25836fcdd927bc07452c625189eab2a2b9> 	1cd2da8 <https://github.com/al177/esp8089/commit/1cd2da85c1cb32252ee150f77bd96573d8ea980c>
> Firmware built-in (.h) support 	YES 	YES 	NO, dropped during upstreaming attempt
> Firmware in eagle_fw{1,2,3}.h 	YES 	YES 	YES (should really be dropped)
> Firmware .bin files in repo 	NO 	jwrdegoede/esp8089@4445df9 <https://github.com/jwrdegoede/esp8089/commit/4445df9c63be5d92268d0304b2adfeb723c9f87f> 	Icenowy/esp8089@711334e <https://github.com/Icenowy/esp8089/commit/711334e24668d4660e43d4ad2e3cf8c054b7a551>
> |MODULE_FIRMWARE()| documentation 	NO 	NO 	NO
> Dropped support for not |P2P_CONCURRENT| 	NO 	NO 	Icenowy/esp8089@711334e <https://github.com/Icenowy/esp8089/commit/711334e24668d4660e43d4ad2e3cf8c054b7a551>
> Dropped support for |testmode.c| 	NO 	jwrdegoede/esp8089@dd28cb5 <https://github.com/jwrdegoede/esp8089/commit/dd28cb55a1b67c3fc81ef63617eee9c44b6f21ec> 	Icenowy/esp8089@20f0ff0 <https://github.com/Icenowy/esp8089/commit/20f0ff05bf1431f8fc23d0f3bf6b86b91cef82c2>
> Dropped support for |FAST_TX_STATUS| 	NO 	NO 	Icenowy/esp8089@8cd5735 <https://github.com/Icenowy/esp8089/commit/8cd5735bedee11fb1861a880fb16f9693816594a>
> Dropped support for Android, SPI interface 	NO 	NO 	Icenowy/esp8089@909c1f6 <https://github.com/Icenowy/esp8089/commit/909c1f673f16684ed32d2194f225367cedfc510d>, Icenowy/esp8089@93f574b <https://github.com/Icenowy/esp8089/commit/93f574bc689cd125c9f15c93593311bb0d19e91a>
> |esp_reset_gpio| modules parameter 	al177/esp8089@17f726a <https://github.com/al177/esp8089/commit/17f726ab5e14063db0996aabef22cfcf3a2ef653>, al177/esp8089@c01f5bd <https://github.com/al177/esp8089/commit/c01f5bde78c6bc15ff4cf5e2100805be2c76ba2b>, al177/esp8089@879a31f <https://github.com/al177/esp8089/commit/879a31f0e7da0f6d450f171b74ae43f1a2c3f684> 	Dropped in jwrdegoede/esp8089@a9fd9d6 <https://github.com/jwrdegoede/esp8089/commit/a9fd9d6ab40f4c74a3ed6e55901f642e2fd299d1> because Device tree 	NO, support never merged
>
> @Icenowy <https://github.com/Icenowy>: The merge window for 4.8 is currently open. Will you submit a V3 patch? If yes, please consider:
>
>   * Reverting the drop of frame aggregation in Icenowy/esp8089@79351b1 <https://github.com/Icenowy/esp8089/commit/79351b14abaca2915ab2729c758ce337616cfcf8> and apply the fix from jwrdegoede/esp8089@865e748 <https://github.com/jwrdegoede/esp8089/commit/865e748f060f4bcbefc0c16aa786e37551703bf3>
>   * Applying the fix of |#include <../net/mac80211/ieee80211_i.h>| in |esp_mac80211.c| (see al177/esp8089@097a1ef <https://github.com/al177/esp8089/commit/097a1ef0683d09dc29eb57a22a53fb49794a5fab>)
>   * Add |MODULE_FIRMWARE()| macros so the required firmwares are listed properly by modinfo
>   * Drop |eagle_fw1.h|, |eagle_fw2.h| and |eagle_fw3.h| since they are no longer needed with external firmware files
[... cut ...]
>   * Drop code guarded by |#ifdef TEST_MODE|

Note that my version of the driver currently does not work! It is still a work in progress,
I believe that the problem is that I dropped the reset-gpio. It seems that there is both
an enable (handle through mmc-pwrseq in my code) and a reset gpio (unhandled in my code)
and that the reset is needed to switch from one firmware to the other. At least that is
my theory, I've not continued work on this due to lack of time. I will likely pick this
up again at some point in time.

Regards,

Hans

Icenowy, 2016-07-29:
You can make reset-gpio firstly a device-tree-work and use DT fixed-regulator to pull it up. It's how RTL8723BS is driven on CHIP.

Note that my version of the driver currently does not work! It is still a work in progress,
I believe that the problem is that I dropped the reset-gpio. It seems that there is both
an enable (handle through mmc-pwrseq in my code) and a reset gpio (unhandled in my code)
and that the reset is needed to switch from one firmware to the other. At least that is
my theory, I've not continued work on this due to lack of time. I will likely pick this
up again at some point in time.

Regards,

Hans

jwrdegoede, 2016-07-29:
Hi,

> You can make reset-gpio firstly a device-tree-work and use DT fixed-regulator to pull it up. It's how RTL8723BS is driven on CHIP.

I'm doing that (using mmc-pwrseq, which really is the preferred way to do this),
but I get some timeout errors half way through probing the esp8089, as said
I've not yet had time to debug this further.

On what boards have you tested your version of the esp8089 driver ?

Do you've a dts example ?

Regards,

Hans

Icenowy, 2016-07-29:
I tested it on my A33 Q8 Tablet. (Aoson M751s)
In Icenowy/esp8089 there's a dt example.

> You can make reset-gpio firstly a device-tree-work and use DT fixed-regulator to pull it up. It's how RTL8723BS is driven on CHIP.

I'm doing that (using mmc-pwrseq, which really is the preferred way to do this),
but I get some timeout errors half way through probing the esp8089, as said
I've not yet had time to debug this further.

On what boards have you tested your version of the esp8089 driver ?

Do you've a dts example ?

Regards,

Hans

jwrdegoede, 2016-07-29:
Hi,

> I tested it on my A33 Q8 Tablet. (Aoson M751s)
> In Icenowy/esp8089 there's a dt example.

Cool, if you've this working on a q8 tablet, I guess I should be
able to get it to work too. I'll give your code a try. Could you
add the firmware files you're using to your git repo ?

Regards,

Hans

Icenowy, 2016-07-29:
>
> Hi,
>
> > I tested it on my A33 Q8 Tablet. (Aoson M751s)
> > In Icenowy/esp8089 there's a dt example.
>
> Cool, if you've this working on a q8 tablet, I guess I should be
> able to get it to work too. I'll give your code a try. Could you
> add the firmware files you're using to your git repo ?
they are at "cleanup" branch of my repo.
(This branch is more mainline-friendly...)

noahwilliamsson, 2016-08-01:
FWIW, I've successfully built and tested the cleanup branch on a 4.7 kernel (https://github.com/raspberrypi/linux/tree/rpi-4.7.y) with an ESP-12F chip hooked up to a Raspberry Pi Zero via SDIO, as outlined on https://hackaday.io/project/8678-rpi-wifi#menu-instructions.

To get it to work I had to change `INIT_DATA_CONF_BUF` (_esp_conf.h_) and set `crystal_26M_en=1`.  It seems that you set to it zero in Icenowy/esp8089@df40d02d525e6dc529c34942814796b9e538a602.  Doesn't WiFi work on your hardware with this setting enabled?

Before making that change WiFi didn't work at all for me:  `wpa_cli scan_results` always returned an empty list of found WiFi networks.  I'll also note that the debugging output on the chip's UART didn't come at a standard baud rate (and I couldn't really bother figuring out the exact timing).   With `crystal_26M_en=1` there's usually chatter at 115200 in the form of:
> w
> s
> nOFF, ch 1 0x53000000
> nOFF, ch 1 0x50520000
> w
> nON, ch 1 0x50530000
> s
> nOFF, ch 1 0x50510000

Icenowy, 2016-08-01:
>
> FWIW, I've successfully built and tested the cleanup branch on a 4.7 kernel (https://github.com/raspberrypi/linux/tree/rpi-4.7.y) with an ESP-12F chip hooked up to a Raspberry Pi Zero via SDIO, as outlined on https://hackaday.io/project/8678-rpi-wifi#menu-instructions.
>
> To get it to work I had to change INIT_DATA_CONF_BUF (esp_conf.h) and set crystal_26M_en=1. It seems that you set to it zero in Icenowy/esp8089@df40d02. Doesn't WiFi work on your hardware with this setting enabled?
On my device, set it to 1 will make it not work.
Maybe I should make the conf a firmware. (My device is a esp8089 on a Allwinner A33 tablet)
> Before making that change WiFi didn't work at all for me: wpa_cli scan_results always returned an empty list of found WiFi networks. I'll also note that the debugging output on the chip's UART [...]
My esp chip have no UART...

noahwilliamsson, 2016-08-02:
> On my device, set it to 1 will make it not work.
> Maybe I should make the conf a firmware. (My device is a esp8089 on a Allwinner A33 tablet)

According to http://wenku.baidu.com/view/f78408b19b89680203d82576.html this setting is to select between 40MHz crystals (like on your ESP8089) and 26MHz crystals (like on my ESP8266 ESP-12F).  There was also some discussion on http://www.esp8266.com/viewtopic.php?f=13&t=8119.

I've created a PR on your cleanup branch that:

- applies the INIT_DATA_CONF_BUF for an ESP8089 to the existing config table
- allows the configuration to be changed via a module parameter `config`

On a different note I've tried to revert the commit Icenowy/esp8089@79351b14abaca2915ab2729c758ce337616cfcf8 (where you ripped out support for A-MPDU) and then apply the fix to compile under v4.6 and later.  For reasons I don't yet understand this completely breaks WiFi transmissions and scanning (empty scan result) shortly after authenticating with the AP.  What I see in dmesg is:

```
[   98.617036] esp_op_add_interface STA
[   98.618066] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   99.553640] wlan0: authenticate with 01:aa:bb:cc:dd:ee
[   99.555595] wlan0: send auth to 01:aa:bb:cc:dd:ee (try 1/3)
[   99.578928] wlan0: authenticated
[   99.584946] wlan0: associate with 01:aa:bb:cc:dd:ee (try 1/3)
[   99.612062] wlan0: RX AssocResp from 01:aa:bb:cc:dd:ee (capab=0x1411 status=0 aid=3)
[   99.612241] wlan0: associated
[   99.612759] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   99.619286] esp_op_ampdu_action: enter action 0
[   99.619327] esp_op_ampdu_action RX START 01:aa:bb:cc:dd:ee tid 0 0
[   99.677966] wlan0: Limiting TX power to 20 (20 - 0) dBm as advertised by 01:aa:bb:cc:dd:ee
[   99.871440] esp_op_ampdu_action: enter action 0
[   99.871504] esp_op_ampdu_action RX START 01:aa:bb:cc:dd:ee tid 6 0
[  101.674622] esp_op_ampdu_action: enter action 2
[  101.674664] esp_op_ampdu_action TX START, addr:01:aa:bb:cc:dd:ee,tid:0,state:1
[  101.674678] esp_op_ampdu_action: TX_START ieee80211_start_tx_ba_cb_irqsafe(tid=0)
[  101.688401] esp_op_ampdu_action: enter action 6
[  101.688483] esp_op_ampdu_action TX OPERATION, addr:01:aa:bb:cc:dd:ee,tid:0,state:3
[  101.688502] ESSERT:/home/pi/src/wip-esp8089/esp_mac80211.c 1298
[  105.773878] esp_op_ampdu_action: enter action 4
[  105.773925] esp_op_ampdu_action: enter action 1
[  105.773941] esp_op_ampdu_action RX STOP 01:aa:bb:cc:dd:ee tid 0
[  105.773967] esp_op_ampdu_action: enter action 1
[  105.773978] esp_op_ampdu_action RX STOP 01:aa:bb:cc:dd:ee tid 6
[  106.019748] wlan0: authenticate with 01:02:03:cc:bb:aa
[  106.033013] wlan0: send auth to 01:02:03:cc:bb:aa (try 1/3)
[  106.233899] wlan0: send auth to 01:02:03:cc:bb:aa (try 2/3)
[  106.443898] wlan0: send auth to 01:02:03:cc:bb:aa (try 3/3)
[  106.653814] wlan0: authentication with 01:02:03:cc:bb:aa timed out
[  107.580107] wlan0: authenticate with 01:aa:bb:cc:dd:ee
[  107.580321] wlan0: send auth to 01:aa:bb:cc:dd:ee (try 1/3)
[  107.783857] wlan0: send auth to 01:aa:bb:cc:dd:ee (try 2/3)
[  107.993856] wlan0: send auth to 01:aa:bb:cc:dd:ee (try 3/3)
[  108.203882] wlan0: authentication with 01:aa:bb:cc:dd:ee timed out
```

Interestingly, TX aggregation works fine in the al177/esp8089 driver (with a v4.6+ patch).  Despite going through commit histories and double-checking firmware files I haven't been able to isolate the difference yet.  Here's the corresponding dmesg:

```
[  962.446404] esp_op_add_interface STA
[  962.447409] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  963.376674] wlan0: authenticate with 01:aa:bb:cc:dd:ee
[  963.388250] wlan0: send auth to 01:aa:bb:cc:dd:ee (try 1/3)
[  963.391730] wlan0: authenticated
[  963.401191] wlan0: associate with 01:aa:bb:cc:dd:ee (try 1/3)
[  963.404903] wlan0: RX AssocResp from 01:aa:bb:cc:dd:ee (capab=0x1411 status=0 aid=3)
[  963.405027] wlan0: associated
[  963.405397] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  963.414177] esp_op_ampdu_action: enter action 0
[  963.414208] esp_op_ampdu_action RX START 01:aa:bb:cc:dd:ee tid 0 0
[  963.437837] wlan0: Limiting TX power to 20 (20 - 0) dBm as advertised by 01:aa:bb:cc:dd:ee
[  964.032667] esp_op_ampdu_action: enter action 0
[  964.032698] esp_op_ampdu_action RX START 01:aa:bb:cc:dd:ee tid 6 0
[  965.085535] start tx ba session,addr:01:aa:bb:cc:dd:ee,tid:0
[  965.085799] esp_op_ampdu_action: enter action 2
[  965.085827] esp_op_ampdu_action TX START, addr:01:aa:bb:cc:dd:ee,tid:0,state:1
[  965.085841] esp_op_ampdu_action: TX_START ieee80211_start_tx_ba_cb_irqsafe(tid=0)
[  965.087798] esp_op_ampdu_action: enter action 6
[  965.087832] esp_op_ampdu_action TX OPERATION, addr:01:aa:bb:cc:dd:ee,tid:0,state:2
```

Icenowy, 2016-08-02:
yes it's strange.
But without amsdu it can still work...

On a different note I've tried to revert the commit Icenowy/esp8089@79351b1 (where you ripped out support for A-MPDU) and then apply the fix to compile under v4.6 and later.  For reasons I don't yet understand this completely breaks WiFi transmissions and scanning (empty scan result) shortly after authenticating with the AP.  What I see in dmesg is:

zoobab, 2019-01-29:
Any status on the upstream attempt?

noahwilliamsson, 2022-04-13:
@zoobab No updates since an attempt in 2017 according to https://lore.kernel.org/lkml/?q=esp8089.

From what I can tell there are device tree bindings for the esp8089 but they were merged with a reference to an out-of-tree driver: https://github.com/torvalds/linux/search?p=1&q=esp8089&type=commits.

Thanks for the interest, everyone.  Closing.

closed 2022-04-13 by noahwilliamsson