al177/esp8089 #6: AP Mode Supported?

jextrevor, 2016-11-27:
In its current state, does this driver support AP (Access Point) mode? That is, when using this driver, can one set up a wireless access point?

If not, what work is required to implement this functionality?

al177, 2016-11-29:
Yes, it works in AP mode.  Install this driver, then follow any of the guides for getting a Raspberry Pi with USB WiFi or Pi 3 WiFi set up as an AP.

closed 2016-11-29 by al177

jextrevor, 2016-11-29:
Much thanks.

haoning747, 2019-01-06:
@jextrevor  Do you successfully setup AP mode? Can you give me some direction or learning materials?

jextrevor, 2019-01-06:
Unfortunately I haven't set it up because I don't have a physical copy of
the hardware.

haoning747, 2019-01-06:
> Unfortunately I haven't set it up because I don't have a physical copy of the hardware.
> […](#)

I hava spent much time on  this problem, some guys in China say it's very easy but I need to pay money...It seems only money can solve it.

CRImier, 2019-01-06:
> some guys in China say it's very easy but I need to pay money

That's now how this works. You should be able to follow some hostapd-based tutorial on the Internet - google for "raspberry pi hostapd". Any problems you encounter on the way probably won't be hardware-specific and will be solvable (which is to say - don't bug this repo about your hostapd problems unless you're *sure* the problem is because of this driver). You don't need to pay money for this - not that anybody can stop you, but it wouldn't be a wise investment.

haoning747, 2019-01-06:
> raspberry pi hostapd

Sorry, Your "big google" has been shield by our government...It's time to use VPN, thx.

CRImier, 2019-01-06:
Ah, that's understandable. Happy to hear that at least VPN works for you! If not, I'm sure your local search engine can provide some tutorials if you look up "hostapd" specifically.