al177/esp8089 #7: 'no .config found' on install?

czarina, 2017-02-01:
This is the error I get on 'make sudo make install':

Makefile:19: *** No .config found in /lib/modules/4.4.34+/build, please set KBUILD to configured kernel.  Stop.

al177, 2017-02-03:
The ideal way to install this is via DKMS, which will handle building the module against the current kernel.  Follow the directions here: [https://hackaday.io/project/8678/instructions](https://hackaday.io/project/8678/instructions)

If you want to build directly from this git repo, you'll need to install the matching kernel headers, which Raspbian doesn't provide in package form.  [rpi-source is a good tool](https://github.com/notro/rpi-source/wiki) for fetching and configuring the source tree for the currently running kernel.

al177, 2017-02-05:
@czarina : Please update this issue when youve had a chance to try one of the above suggestions.  If the KBUILD process is broken due to some change in Raspbian I'd like to fix it.

closed 2017-02-13 by al177