al177/esp8089 #11: Raspberry Pi Compute Module 3 / Jessie Lite

keabarnes, 2017-04-11:
I've made multiple attempts to get this up and running and can't seem to get it right.

I'd just like to confirm, is there any reason why this would not work on a Raspberry Pi Compute Module 3 running the latest Jessie Lite?

I'm getting the -110 error on boot and there's occasionally a kernal panic dump even when I'm not doing anything.

I'll be trying another fresh install tomorrow using the instructions on 
https://hackaday.io/project/8678/instructions

As far as I can tell, I don't think it's the ESP8266 module, I've connected the module to a RPi 1 Model B and it worked.

keabarnes, 2017-04-12:
For reference, my dmesg after boot on  a fresh install of "2017-04-10-raspbian-jessie-lite" on CM3 (with eMMC option):
[dmesg.txt](https://github.com/al177/esp8089/files/916033/dmesg.txt)

al177, 2017-04-16:
I have not tested on BCM2837, and I do know the RPi3 has the SDIO controller mapped to different pins by default for the onboard WiFi.

al177, 2017-04-17:
Looking at your logs, the ESP is showing up on the SDIO bus (at least the SD_CMD is), so it's not a port routing issue.  One problem I've discovered along the way is that the CH_PD signal pullup on the ESP8266 is inconsistent from chip to chip.

Try putting a 4.7K pullup resistor between CH_PD and 3.3V and see if this goes away.

Failing that, you'll need to scope the SDIO signals at the module and see if there's a decent eye.  You may need to add or remove termination resistance, or you might have a wiring problem.  SD/SDIO is difficult in that the device can respond properly to a probe as long as SD_CS, SD_CMD, and SD_CLK are correctly connected, even if SD_D[3:0] are not.

keabarnes, 2017-04-20:
Thanks for taking the time to reply. I'd read that about the CH_PD pull-up as well and have already added a 3.9k resistor to 3v3, which unfortunately hasn't helped. 

Hooking up the scope to check the eye is a good idea, I'll try that out. Do you know if the SD_D[3:0] lines are active during the esp8089 driver load stage?

al177, 2017-04-20:
For the SD bus, SD_CMD carries device probe requests/responses, commands to the device, and device ID.  SD_D[3:0] is purely data going to/from the device.  If you think about it like a classic A/D bus, SD_CMD is address (ignoring the out-of-band stuff) and SD_D[3:0] is data.

When the firmware load happens, SD_CMD sets up the card to receive firmware, then the firmware is sent across SD_D[3:0].

closed 2017-05-13 by al177