al177/esp8089 #16: Compile Error on 4.9.50-1-ARCH

SPuntte, 2017-09-17:
I was fiddling with this, [so far unsuccesfully](https://imgur.com/a/Q7ZeI), on a Raspberry Pi Model B Rev 2 (0x000f) running Arch Linux ARM (kernel 4.9.50-1-ARCH) and encountered a breaking change in `include/net/mac80211.h` API. It seems the change was introduced in 4.6. (Even though #14 claims compatibility with 4.9?)

```
$ make
make -C /usr/lib/modules/4.9.50-1-ARCH/build M=/home/alarm/build/esp8089
make[1]: Entering directory '/usr/lib/modules/4.9.50-1-ARCH/build'
  CC [M]  /home/alarm/build/esp8089/esp_mac80211.o
/home/alarm/build/esp8089/esp_mac80211.c: In function 'hw_scan_done':
/home/alarm/build/esp8089/esp_mac80211.c:900:9: warning: ISO C90 forbids mixed declarations and code [-Wdeclaration-after-statement]
         struct cfg80211_scan_info info = {
         ^~~~~~
/home/alarm/build/esp8089/esp_mac80211.c: In function 'hw_scan_timeout_report':
/home/alarm/build/esp8089/esp_mac80211.c:934:9: warning: ISO C90 forbids mixed declarations and code [-Wdeclaration-after-statement]
         struct cfg80211_scan_info info = {
         ^~~~~~
/home/alarm/build/esp8089/esp_mac80211.c: At top level:
/home/alarm/build/esp8089/esp_mac80211.c:1884:25: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
         .ampdu_action = esp_op_ampdu_action,
                         ^~~~~~~~~~~~~~~~~~~
/home/alarm/build/esp8089/esp_mac80211.c:1884:25: note: (near initialization for 'esp_mac80211_ops.ampdu_action')
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:294: /home/alarm/build/esp8089/esp_mac80211.o] Error 1
make[1]: *** [Makefile:1493: _module_/home/alarm/build/esp8089] Error 2
make[1]: Leaving directory '/usr/lib/modules/4.9.50-1-ARCH/build'
make: *** [Makefile:91: modules] Error 2
```

I have little time and experience with kernel(module) hacking so I'll just leave a naïve fix here if someone stumbles upon the same issue or wants to refine this into an actual patch.
```diff
diff --git a/esp_mac80211.c b/esp_mac80211.c
index 9403e5a..6345b13 100755
--- a/esp_mac80211.c
+++ b/esp_mac80211.c
@@ -1569,15 +1569,29 @@ static int esp_op_ampdu_action(struct ieee80211_hw *hw,
                                struct ieee80211_sta *sta, u16 tid, u16 *ssn,
                                u8 buf_size)
 #else
+#if (LINUX_VERSION_CODE < KERNEL_VERSION(4, 6, 0))
 static int esp_op_ampdu_action(struct ieee80211_hw *hw,
                                struct ieee80211_vif *vif,
                                enum ieee80211_ampdu_mlme_action action,
                                struct ieee80211_sta *sta, u16 tid, u16 *ssn,
                                u8 buf_size, bool amsdu)
+#else
+static int esp_op_ampdu_action(struct ieee80211_hw *hw,
+                               struct ieee80211_vif *vif,
+                              struct ieee80211_ampdu_params *params)
+#endif
 #endif
 #endif
 #endif /* NEW_KERNEL && KERNEL_35 */
 {
+#if (LINUX_VERSION_CODE >= KERNEL_VERSION(4, 6, 0))
+       enum ieee80211_ampdu_mlme_action action = params->action;
+       struct ieee80211_sta *sta = params->sta;
+       u16 tid = params->tid;
+       u16 *ssn = &params->ssn;
+       u8 buf_size = params->buf_size;
+       //bool amsdu = params->amsdu;
+#endif
         int ret = -EOPNOTSUPP;
         struct esp_pub *epub = (struct esp_pub *)hw->priv;
 #if (LINUX_VERSION_CODE >= KERNEL_VERSION(2, 6, 28))
```

al177, 2017-09-19:
That strange that it hasn't broken builds on 4.9 Raspbian.  I'll try to do a fresh build with the latest Raspbian tonight, and then see if jwrdegoede's patch causes any issues.

CRImier, 2017-11-26:
Can confirm that it can't be installed on latest Raspbian =(

CRImier, 2017-11-27:
So, [I've compiled and released](https://github.com/ZeroPhone/esp8089/releases/tag/1.9.20171127) a new package, containing the aforementioned fix. It seems to work so far, though I don't understand the code enough to be able to tell if anything could've broken.

al177, 2017-12-24:
#18 should have fixed the current kernel builds with Raspbian.

closed 2017-12-24 by al177