al177/esp8089 #19: compilation impossible in latest Raspbian, KBUILD is missing

voloviq, 2017-12-18:
Hi,
I try to compile the latest esp8089 sdio driver in latest raspbian but without success. I got a following error

"Makefile:14: *** Kernel build tree not found - please set KBUILD to configured kernel. Stop."

How can I set KBUILD

Thanks

al177, 2017-12-18:
The Makefile guesses at the kernel header path by looking for the /lib/modules/$(RUNNING_VERSION)/build .  The raspberrypi-kernel-headers package should automatically set up a symlink.  If you've built your own kernel, you will need to set KBUILD in the shell: `KBUILD=/usr/src/my_linux-4.9.0 make all`

If you're using the Raspbian packaged kernels, I recommend installing from the DKMS source package in the releases section of this repo or using the `make dkms` target to register and build this driver with DKMS.  You'll need to install packages "dkms" and "raspberrypi-kernel-headers" first.

voloviq, 2017-12-19:
Hi,
thank for the answer. After investigating Makefile and read Your answer above I check my file structure. Of course, kernel headers were missing. So I simply install it by issue sudo apt-get install raspberrypi-kernel-headers on the command line. So now driver compile as it should. I try to use Your code into bbb Yocto, it will be a bit different but I hope to succeed.
I think that this issue can be close.
Thanks
BR
Voloviq

closed 2017-12-19 by voloviq