al177/esp8089 #20: sdmmc_ack_interrupt missing on TI kernel 4.9.24

voloviq, 2017-12-19:
Hello,
I try to port esp8089 driver source code to bbb kernel 4.9.24. So I get two following errors

1. in file testmode.c in line 985

      rc = genl_register_family_with_ops(&test_genl_family,
                                   esp_test_ops, ARRAY_SIZE(esp_test_ops));

I modified to 
     rc = genl_register_family_with_ops(&test_genl_family, esp_test_ops);

because in file genetlink.h I found a modified macro

#define genl_register_family_with_ops(family, ops)			\
	_genl_register_family_with_ops_grps((family),			\
					    (ops), ARRAY_SIZE(ops),	\
					    NULL, 0)

from C language code is consistent due to size calculation is made in _genl_register_family_with_ops_grps(...)

So I think that modifcation no change code to much and should work well. I hope You agree with me regarding this point. Please confirm.

2. In file stdio_stub.c there is lack of "sdmmc_ack_interrupt(...) function body. In line 86 exist invocation of mentioned function. During compilation, linker assign me an error due to function body not found anywhere.

Orignal message from compilation window
"
| drivers/built-in.o: In function `sif_platform_ack_interrupt':
| core.c:(.text+0x1a08a4): undefined reference to `sdmmc_ack_interrupt'
| make[2]: *** [vmlinux] Error 1
| make[1]: *** [sub-make] Error 2
| make: *** [__sub-make] Error 2
| ERROR: oe_runmake failed
| WARNING: /home/mw/yocto/tmp/work/am335x-poky-linux-gnueabi/ti-linux/4.9.24-r4/temp/run.do_compile.32589:1 exit 1 from 'exit 1'
"
My question is can We somehow workaround this issue. Maybe exist similar function with appropriate functionality? Maybe I can disable definition of **ESP_ACK_INTERRUPT** at all?

At the end, I can write that when I comment line sdmmc_ack_interrupt(func->card->host); (line number 86 in sdio_stub.c file) all source code compile well and generate *.ko file.

Thanks in advanced
Best Regards
Voloviq

voloviq, 2017-12-19:
Hi,

I do not see exactly in Makefile but when I copied following flags

EXTRA_CFLAGS += -DDEBUG -DSIP_DEBUG -DFAST_TX_STATUS -DKERNEL_IV_WAR -DRX_SENDUP_SYNC -DDEBUGFS -DTEST_MODE -DHAS_INIT_DATA -DHAS_FW

EXTRA_CFLAGS += -DP2P_CONCURRENT -DESP_USE_SDIO

ifdef ANDROID
EXTRA_CFLAGS += -DANDROID
endif

ifdef P2P_CONCURRENT
EXTRA_CFLAGS += -DP2P_CONCURRENT
endif

ifdef TEST_MODE
EXTRA_CFLAGS += -DTEST_MODE
endif

during compilation function void sif_platform_ack_interrupt(struct esp_pub *epub) is not taken to compilation.

Please confirm my only point number 1 and issue can be closed.

al177, 2017-12-24:
I'm not sure how you were building previously, but you definitely need those EXTRA_CFLAGS defined when building against a kernel.  I would guess that somehow you were picking up -DTEST_MODE, which isn't needed for normal functioning.

voloviq, 2017-12-29:
Hi Andrew,
I think the building is succeeded now. Currently, I have a bit different problem, the driver seems to work but I don't know how to relate it to dts. The only what I see now is RETRY 3 times and unload finally. I changed reset pin to appropriate for my purpose but what about sdio interface, it is defined as a constant into a code. If yes can You please give me a hint where I can locate it.
Thanks

voloviq, 2017-12-29:
Hi Andrew,
so finally I decided to switch back to Raspberrypi 2 model B v1.1. I compiled everything well with one small warning, sudo make install without any troubles. I made connection as on the following picture 

[image]

I checked connection tiwces.
And the result is as follow

[  117.713263] esp8089: loading out-of-tree module taints kernel.
[  117.718391] 
               ***** EAGLE DRIVER VER:bdf5087c3deb*****

[  117.718499] ESP8089 reset via GPIO 0
[  129.126524] esp_sdio_init ------ RETRY ------ 
[  129.126680] ESP8089 reset via GPIO 0
[  129.346550] ESP8089 reset via GPIO 0
[  140.650138] esp_sdio_init ------ RETRY ------ 
[  140.650295] ESP8089 reset via GPIO 0
[  140.869901] ESP8089 reset via GPIO 0
[  152.174108] esp_sdio_init ------ RETRY ------ 
[  152.174255] ESP8089 reset via GPIO 0
[  152.394206] ESP8089 reset via GPIO 0
[  163.696191] esp_sdio_init ------ RETRY ------ 
[  163.696337] ESP8089 reset via GPIO 0
[  163.916229] eagle sdio can not power up!

What I see on wifi module if the blue LED blinks when appears RETRY on screen, so reset works well I guess.
Do You have some recipe for this issue? I can confirm that on bbb I see the same.

voloviq, 2017-12-29:
Hi Andrew,
small correction, to proper work I need to reconfigure sdio interface. So to summarize the following action need to be taken

sudo nano /boot/config.txt
and add in it
dtoverlay=sdio,poll_once=off
finall reboot.

voloviq, 2017-12-29:
I think I'm not far away to close this issue, please confirm me only if sdio number 3 interface and 4 bit sdio mode is hardcoded. If yes please write in which files I should looking for and apply changes.

Thanks

al177, 2017-12-30:
The SDIO interface isn't hardcoded.  The SD host driver probes the device, then the device driver module is loaded based on the probed CID.

The ESP-12F module can run in 1 or 4 bit mode, autonegotiated when the device probes.  For the RPi, the bus_width= parameter forces the driver to 1 bit mode on the host side if you don't wire through SD_D3 and SD_D4.  On any other 4 bit host with only SD_D0 and SD_D1 wired you'll need to do something similar.

voloviq, 2017-12-30:
Thanks Andrew,
I'm curious now only to dts and especially to linies which I need to add
dtoverlay=sdio,poll_once=off
How should look my dts. I'm not able to find sdio.dts but only compiled sdio.dtbo. So how should I configure lines on bbb and which driver use or not use.
My dts looks like

	wifi_pins: pinmux_wifi_card {
		pinctrl-single,pins = <
			AM33XX_IOPAD(0x820, PIN_INPUT | MUX_MODE2) /* GPMC_AD8.mmc1_dat0  */
			AM33XX_IOPAD(0x824, PIN_INPUT | MUX_MODE2) /* GPMC_AD9.mmc1_dat1  */
			AM33XX_IOPAD(0x828, PIN_INPUT | MUX_MODE2) /* GPMC_AD10.mmc1_dat2 */
			AM33XX_IOPAD(0x82C, PIN_INPUT | MUX_MODE2) /* GPMC_AD11.mmc1_dat3 */
			AM33XX_IOPAD(0x880, PIN_INPUT | MUX_MODE2) /* GPMC_CSN1.mmc1_clk  */
			AM33XX_IOPAD(0x884, PIN_INPUT | MUX_MODE2) /* GPMC_CSN2.mmc1_cmd  */
		>;
	};
	
    wifi_rst_pin: pinmux_wifi_rst {
		pinctrl-single,pins = <
			AM33XX_IOPAD(0x960, PIN_INPUT | MUX_MODE7) /* GPMC_AD12.gpio1_12  */
		>;
	};

/* WIFI module */
&mmc1 {
    #address-cells = <1>;
    #size-cells = <0>;

	pinctrl-names = "default";
	pinctrl-0 = <&wifi_pins>;
	vmmc-supply = <&ldo4_reg>;
	mmc-pwrseq = <&wifi_pwrseq>;
	bus-width = <4>;
	non-removable;
	disable-wp;
	status = "okay";
	
	esp8089: sdio_wifi@1 {
        compatible = "esp,esp8089";
        reg = <1>;
        esp,crystal-26M-en = <2>;
    	status = "okay";
    };
}; 

Maybe I do someting wrong.

al177, 2017-12-31:
I can't comment on the SDIO drivers for BBB.  The first step is to see if the SDIO driver is probing the card correctly.  Look at dmesg when the SDIO module loads to see if it finds a device.

voloviq, 2018-01-04:
Hi Andrew,
Yes You right, AM335x don't probe devices correctly. I think that driver can't probe something which not exit. I think this issue can be close, because of it other problem.
Thanks a lot.

closed 2018-01-04 by voloviq

al177, 2018-01-04:
Sorry this isn't working out.  Probing is required for esp8089 re-enumeration after the firmware is loaded.

voloviq, 2018-01-04:
Sorry, I think that I don't understand You correctly. So if I good understand sdio driver itself should discover device(eg. esp8089) on sdio interface before esp8089.ko load. Am I right now? When I give dmesg | grep MMC I see that driver constantly try to find sd/MMC module and try to find wp and cd pins which are not declared into DTS. Tomorrow I try to connect an oscilloscope to see if something happens on sdio pins. I have a strangle filling that it is silent. In by the way On Raspberrypi I was able to see in /sys/kernel/debug two MMC modules on BBB I don't see MMC files.

voloviq, 2018-01-05:
After measurements, I can see on oscilloscope some good waveform on CMD and CLK. Unfortunately, I get something like following after checking what's happen on MMC module

dmesg | grep mmc
[    1.275320] omap_hsmmc 481d8000.mmc: GPIO lookup for consumer cd
[    1.275332] omap_hsmmc 481d8000.mmc: using device tree for GPIO lookup
[    1.275345] of_get_named_gpiod_flags: can't parse 'cd-gpios' property of node '/ocp/mmc@481d8000[0]'
[    1.275354] of_get_named_gpiod_flags: can't parse 'cd-gpio' property of node '/ocp/mmc@481d8000[0]'
[    1.275362] omap_hsmmc 481d8000.mmc: using lookup tables for GPIO lookup
[    1.275372] omap_hsmmc 481d8000.mmc: lookup for GPIO cd failed
[    1.275382] omap_hsmmc 481d8000.mmc: GPIO lookup for consumer wp
[    1.275389] omap_hsmmc 481d8000.mmc: using device tree for GPIO lookup
[    1.275397] of_get_named_gpiod_flags: can't parse 'wp-gpios' property of node '/ocp/mmc@481d8000[0]'
[    1.275405] of_get_named_gpiod_flags: can't parse 'wp-gpio' property of node '/ocp/mmc@481d8000[0]'
[    1.275412] omap_hsmmc 481d8000.mmc: using lookup tables for GPIO lookup
[    1.275420] omap_hsmmc 481d8000.mmc: lookup for GPIO wp failed
[    1.375642] mmc0: queuing unknown CIS tuple 0x01 (3 bytes)
[    1.446847] mmc0: queuing unknown CIS tuple 0x1a (5 bytes)
[    1.461157] mmc0: queuing unknown CIS tuple 0x1b (8 bytes)
[    1.499387] mmc0: error -110 whilst initialising SDIO card
[    2.022957] vmmcsd_fixed: disabling

It seems that MMC driver start to probe but get some strangle answer and finally disable voltage. Any idea what can be wrong?

during insmod I get 

[   72.905483] 
[   72.905483] ***** EAGLE DRIVER VER:bdf5087c3deb*****
[   72.905483] 
[   72.917412] ESP8089 reset via GPIO 44
[   84.328179] esp_sdio_init ------ RETRY ------ 
[   84.334424] ESP8089 reset via GPIO 44
[   84.558263] ESP8089 reset via GPIO 44
[   95.848177] esp_sdio_init ------ RETRY ------ 
[   95.854379] ESP8089 reset via GPIO 44
[   96.078295] ESP8089 reset via GPIO 44
[  107.368180] esp_sdio_init ------ RETRY ------ 
[  107.377586] ESP8089 reset via GPIO 44
[  107.598220] ESP8089 reset via GPIO 44
[  118.888176] esp_sdio_init ------ RETRY ------ 
[  118.894407] ESP8089 reset via GPIO 44
[  119.118225] eagle sdio can not power up!
insmod: ERROR: could not insert module wlan.ko: No such device

of course I have to change reset pin. Reset pin works fine I see short blue LED blinking during each probe

voloviq, 2018-01-05:
Now I think I really close. Please take a look

[    4.388005] esp_sdio_dummy_probe enter
[    4.618273] esp_sdio_init power up OK
[    5.128028] esp_download_fw Failed to write fw, err: -84
[    5.190921] esp_init_all failed: -84

It means that some firmware is missing.

voloviq, 2018-01-05:
Finally I driver initialized

[    5.605614] esp_sdio_dummy_probe enter
[    5.818237] esp_sdio_init power up OK
[    7.108605] esp_host:bdf5087c3deb
[    7.108605] esp_target: e826c2b3c9fd 57 18202
[    7.117868] esp_readwrite_file: file /system/lib/modules/test_results filp_open error
[    7.250992] esp_sdio_remove enter
[    7.948565] esp_host:bdf5087c3deb
[    7.948565] esp_target: e826c2b3c9fd 57 18202

but when I add 
ifup wlan0

Successfully initialized wpa_supplicant
[   86.026168] esp_op_add_interface STA 
[   86.030942] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
ioctl[SIOCSIWPMKSA]: Invalid argument
ioctl[SIOCSIWMODE]: Invalid argument
ioctl[SIOCGIWRANGE]: Invalid argument
ioctl[SIOCGIWMODE]: Invalid argument
ioctl[SIOCSIWAP]: Invalid argument
ioctl[SIOCSIWESSID]: Invalid argument
ioctl[SIOCSIWENCODEEXT]: Invalid argument
ioctl[SIOCSIWENCODEEXT]: Invalid argument
ioctl[SIOCSIWENCODEEXT]: Invalid argument
ioctl[SIOCSIWENCODEEXT]: Invalid argument
ioctl[SIOCSIWPMKSA]: Invalid argument
udhcpc (v1.24.1) started
Sending discover...
Sending discover...
Sending discover...
No lease, forking to background

No I think that some other driver are missing. Maybe You known which?

voloviq, 2018-01-05:
Of course after issue ifconfig I get

wlan0     Link encap:Ethernet  HWaddr 18:FE:34:34:93:AD  
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

voloviq, 2018-01-05:
After changing a few options in kernel wifi works as it should. Now issue can be closed. In free time I made a fork and describe what is necessary to run this driver on Am335x.

Thanks a lot

al177, 2018-01-05:
Excellent!