al177/esp8089 #21: esp8089 chip won't works with wlan

voloviq, 2018-01-23:
Hi Andrew,
I just finalize working with ESP WiFi modules and I have a bit rouble with IC esp8089. On the same condition chip esp8266 works very well. I don't know why.
Here I have a driver info during load.
So for 
ESP8089
[    5.362510] esp_sdio_dummy_probe enter
[    5.598617] esp_sdio_init power up OK
[    6.778174] esp_host:bdf5087c3deb
[    6.778174] esp_target: e826c2b3c9fd 57 18202
[    6.787452] esp_readwrite_file: file /system/lib/modules/test_results filp_open error
[    6.825731] esp_sdio_remove enter
[    7.488414] esp_host:bdf5087c3deb
[    7.488414] esp_target: e826c2b3c9fd 57 18202
[    9.030310] esp_op_add_interface STA

ESP8266
[    5.409483] esp_sdio_dummy_probe enter
[    5.628760] esp_sdio_init power up OK
[    6.778176] esp_host:bdf5087c3deb
[    6.778176] esp_target: e826c2b3c9fd 57 18202
[    6.787454] esp_readwrite_file: file /system/lib/modules/test_results filp_open error
[    6.825552] esp_sdio_remove enter
[    7.488397] esp_host:bdf5087c3deb
[    7.488397] esp_target: e826c2b3c9fd 57 18202
[    9.007358] esp_op_add_interface STA 

at the end when 

ifup wlan0 (esp8089)
Successfully initialized wpa_supplicant
[  103.100768] esp_op_add_interface STA 
[  103.105241] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
udhcpc (v1.24.1) started
Sending discover...
Sending discover...
Sending discover...
No lease, forking to background

ifup wlan0 (esp8266)
Successfully initialized wpa_supplicant
[  374.898611] esp_op_add_interface STA 
[  374.903070] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
udhcpc (v1.24.1) started
Sending discover...
[  376.200589] wlan0: authenticate with xxxx
[  376.206614] wlan0: send auth to xxxx (try 1/3)
[  376.228304] wlan0: authenticated
[  376.238416] wlan0: associate with xxxx(try 1/3)
[  376.258262] wlan0: RX AssocResp from xxxx (capab=0x431 status=0 aid=7)
[  376.266167] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  376.273583] wlan0: associated
[  376.398231] esp_op_ampdu_action RX START xxxx tid 0 0
[  377.881760] start tx ba session,addr:xxx,tid:0
[  377.917998] esp_op_ampdu_action TX START, addr:xxx,tid:0,state:1
[  377.948173] esp_op_ampdu_action TX OPERATION, addr:xxxx,tid:0,state:2
Sending discover...
Sending select for 192.168.1.217...
Lease of 192.168.1.217 obtained, lease time 10800
/etc/udhcpc.d/50default: Adding DNS 192.168.1.1

What can be wrong. Maybe firmware (magic arrays with firmware in *.h) files. Can You give me some hints for that.

Thanks
Best Regards

al177, 2018-01-23:
According to an Espressif employee, ESP8089 and ESP8266 are identical except that ESP8266 has additional functionality in ROM (addl boot mode related stuff and wifi HAL library calls).  When using the driver firmware loaded into SRAM, ESP8089 and ESP8266 are functionally identical.

I suspect that there's some hardware difference between your ESP8089 and ESP8266 platforms aside from the IC itself.  So let's drill down into that.

First check that it can see APs at all.  Does "iwlist scan" return any access points at all?  Does it return about as many APs as the ESP8266 in the same location?

Next look at the crystal.  This driver assumes a 26MHz crystal.  If the ESP8089 board is not using a 26MHz crystal then you'll have to change the flag in esp_conf.h to match.

Do you have a schematic?  Can you examine the routing to the VDD* signals?  ESP8089/8266 has particular requirements for routing and decoupling the various power supply inputs.

voloviq, 2018-01-23:
Hi Andrew,

I made my own hardware. I use the same board, the same resonator 26MHz for both chip. In Linux wlan0 appears but when I issue iwlist wlan0 scan system inform me that module not support scanning. The driver is not associated with wlan. 

It is my schematic

[image]

This schematic is based on application from esp8089. The only one difference between esp8089 and eesp8266 is pin MTDO. This pin has to be tied to VDD for esp8266. I try also use serial resistor 200Ohm in SDIO, CLK and CMD lines but also without difference. Of course I disconneced or connected for esp8089 but without any diference. Do You have any ide what can be wrong and where to start find a problem.

al177, 2018-01-23:
Your schematic looks fine.  GPIO0, GPIO2, and MTDO control the boot mode.  If you're getting this far in the driver, then I don't think those are related to the problem.  The fact that the driver makes it past the firmware load is a good sign that your SDIO bus is working.

This driver does support scanning.  You have to "ifconfig wlan0 up" first.

When you say that you are seeing differences between ESP8266 and ESP8089, are you using a different board for ESP8266 or have you soldered an ESP8266 in place of the 8089 in your design?

voloviq, 2018-01-23:
Hi Andrew,
After issue one post  above I realized that I wrong connect pin number 16 to 3.3V is should be NC. I probably damaged my chip. I think it can be an issue. I check it and go back.
Thanks and apologies for a problem.

al177, 2018-01-23:
Not a problem!  Keep this open until you confirm in case there is still a bug.

From the ESP8266 pin list spreadsheet, pin 16 is GPIO4 on the die, which is an input during boot.  I don't know what function it has in ESP8089 firmware, but I don't think you destroyed anything by tying it to 3.3V.  Try cutting the trace and see what happens!

voloviq, 2018-01-24:
Hi Andrew,
I confirm. Pin 16(Vdd for esp8089) must be left unconnect - mandatory. Now WiFi module works exactly like esp8266. Issue can  be closed.

closed 2018-01-24 by voloviq

al177, 2018-01-24:
Nice!