al177/esp8089 #22: Nothing happens after sudo modprobe esp8089

iHD992, 2018-02-04:
Had my RPi Zero with Raspbian Jessie running fine with your driver and my ESP8266.
I decided to install a fresh Raspbian Stretch (no hardware changes).
Everything worked fine until "sudo modprobe esp8089" where just nothing happened:

```
pi@raspberrypi:~ $ wget https://github.com/al177/esp8089/releases/download/1.9.2                                                                                                                                                             0171208/esp8089-dkms_1.9.20171208_all.deb.gz
--2018-02-04 16:48:21--  https://github.com/al177/esp8089/releases/download/1.9.                                                                                                                                                             20171208/esp8089-dkms_1.9.20171208_all.deb.gz
Resolving github.com (github.com)... 192.30.253.112, 192.30.253.113
Connecting to github.com (github.com)|192.30.253.112|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://github-production-release-asset-2e65be.s3.amazonaws.com/475337                                                                                                                                                             17/f549a5a6-dc0e-11e7-9ee4-efd4e16c8429?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-C                                                                                                                                                             redential=AKIAIWNJYAX4CSVEH53A%2F20180204%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-                                                                                                                                                             Date=20180204T154822Z&X-Amz-Expires=300&X-Amz-Signature=97c4043e5603771afd0ffd61                                                                                                                                                             1f137afe2bf35c3d241744ac97232cc6b682e757&X-Amz-SignedHeaders=host&actor_id=0&res                                                                                                                                                             ponse-content-disposition=attachment%3B%20filename%3Desp8089-dkms_1.9.20171208_a                                                                                                                                                             ll.deb.gz&response-content-type=application%2Foctet-stream [following]
--2018-02-04 16:48:22--  https://github-production-release-asset-2e65be.s3.amazo                                                                                                                                                             naws.com/47533717/f549a5a6-dc0e-11e7-9ee4-efd4e16c8429?X-Amz-Algorithm=AWS4-HMAC                                                                                                                                                             -SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20180204%2Fus-east-1%2Fs3%2Faws4                                                                                                                                                             _request&X-Amz-Date=20180204T154822Z&X-Amz-Expires=300&X-Amz-Signature=97c4043e5                                                                                                                                                             603771afd0ffd611f137afe2bf35c3d241744ac97232cc6b682e757&X-Amz-SignedHeaders=host                                                                                                                                                             &actor_id=0&response-content-disposition=attachment%3B%20filename%3Desp8089-dkms                                                                                                                                                             _1.9.20171208_all.deb.gz&response-content-type=application%2Foctet-stream
Resolving github-production-release-asset-2e65be.s3.amazonaws.com (github-produc                                                                                                                                                             tion-release-asset-2e65be.s3.amazonaws.com)... 52.216.83.48
Connecting to github-production-release-asset-2e65be.s3.amazonaws.com (github-pr                                                                                                                                                             oduction-release-asset-2e65be.s3.amazonaws.com)|52.216.83.48|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 2180790 (2.1M) [application/octet-stream]
Saving to: ‘esp8089-dkms_1.9.20171208_all.deb.gz’

esp8089-dkms_1.9.20 100%[===================>]   2.08M  1.17MB/s    in 1.8s

2018-02-04 16:48:25 (1.17 MB/s) - ‘esp8089-dkms_1.9.20171208_all.deb.gz’ saved [                                                                                                                                                             2180790/2180790]

pi@raspberrypi:~ $ gunzip esp8089-dkms_1.9.20171208_all.deb.gz
pi@raspberrypi:~ $ sudo dpkg -i esp8089-dkms_1.9.20171208_all.deb
Selecting previously unselected package esp8089-dkms.
(Reading database ... 155448 files and directories currently installed.)
Preparing to unpack esp8089-dkms_1.9.20171208_all.deb ...
Unpacking esp8089-dkms (1.9.20171208) ...
Setting up esp8089-dkms (1.9.20171208) ...
Loading new esp8089-1.9.20171208 DKMS files...
It is likely that 4.9.59+ belongs to a chroot's host
Building for 4.9.59+ and 4.9.59-v7+
Building for architecture armhf
Building initial module for 4.9.59+
Done.

esp8089:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.9.59+/updates/

depmod.......

DKMS: install completed.
Building initial module for 4.9.59-v7+
Done.

esp8089:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.9.59-v7+/updates/

depmod....

DKMS: install completed.
pi@raspberrypi:~ $ sudo modprobe esp8089
pi@raspberrypi:~ $
```
Same Problem with 1.9.20180103.
Any Ideas? 
I am pretty sure the hardware is ok as it worked with Jessie.

al177, 2018-02-04:
What does "dmesg" say after you install the module?  What about "modprobe"?

iHD992, 2018-02-05:
> What about "modprobe"?
According to your tutorial on hackaday I expect some response.
"If the ESP8089/8266 is found, you should see these lines spread through dmesg: [...]"

> What does "dmesg" say after you install the module?

`Linux raspberrypi 4.9.59+ #1047 Sun Oct 29 11:47:10 GMT 2017 armv6l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Sun Feb  4 17:45:30 2018
pi@raspberrypi:~ $ dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.9.59+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #1047 Sun Oct 29 11:47:10 GMT 2017
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] OF: fdt:Machine model: Raspberry Pi Zero Rev 1.3
[    0.000000] cma: Reserved 8 MiB at 0x17400000
[    0.000000] Memory policy: Data cache writeback
[    0.000000] On node 0 totalpages: 98304
[    0.000000] free_area_init_node: node 0, pgdat c0914e10, node_mem_map d7c8bd00
[    0.000000]   Normal zone: 864 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 98304 pages, LIFO batch:31
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 97440
[    0.000000] Kernel command line: bcm2708_fb.fbwidth=1920 bcm2708_fb.fbheight=1080 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=PARTUUID=b5f06117-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 370984K/393216K available (5950K kernel code, 491K rwdata, 1948K rodata, 396K init, 725K bss, 14040K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xd8800000 - 0xff800000   ( 624 MB)
                   lowmem  : 0xc0000000 - 0xd8000000   ( 384 MB)
                   modules : 0xbf000000 - 0xc0000000   (  16 MB)
                     .text : 0xc0008000 - 0xc05d7a48   (5951 kB)
                     .init : 0xc0841000 - 0xc08a4000   ( 396 kB)
                     .data : 0xc08a4000 - 0xc091ef48   ( 492 kB)
                      .bss : 0xc091ef48 - 0xc09d4648   ( 726 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000032] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000064] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000155] bcm2835: system timer (irq = 27)
[    0.000616] Console: colour dummy device 80x30
[    0.000641] console [tty1] enabled
[    0.000666] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
[    0.060317] pid_max: default: 32768 minimum: 301
[    0.060747] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.060764] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.062010] Disabling memory control group subsystem
[    0.062147] CPU: Testing write buffer coherency: ok
[    0.062202] ftrace: allocating 21715 entries in 64 pages
[    0.180534] Setting up static identity map for 0x8200 - 0x8238
[    0.182511] devtmpfs: initialized
[    0.192284] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
[    0.192710] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.192737] futex hash table entries: 256 (order: -1, 3072 bytes)
[    0.194023] pinctrl core: initialized pinctrl subsystem
[    0.195460] NET: Registered protocol family 16
[    0.197974] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.207629] hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
[    0.207645] hw-breakpoint: maximum watchpoint size is 4 bytes.
[... 193 lines cut ...]
[   85.165120] R8188EU: ERROR assoc success
[   85.175781] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   85.896342] esp_op_add_interface STA
[   85.897149] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
[   87.136705] wlan1: authenticate with c0:25:06:fc:c2:5e
[   87.137874] wlan1: send auth to c0:25:06:fc:c2:5e (try 1/3)
[   87.146205] wlan1: authenticated
[   87.154408] wlan1: associate with c0:25:06:fc:c2:5e (try 1/3)
[   87.161500] wlan1: RX AssocResp from c0:25:06:fc:c2:5e (capab=0x431 status=0 aid=4)
[   87.161626] wlan1: associated
[   87.162012] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[   87.568329] esp_op_ampdu_action RX START c0:25:06:fc:c2:5e tid 0 0
[   88.968863] start tx ba session,addr:c0:25:06:fc:c2:5e,tid:0
[... cut ...]
[   88.975296] esp_op_ampdu_action TX OPERATION, addr:c0:25:06:fc:c2:5e,tid:0,state:2
[   93.946434] Bluetooth: Core ver 2.22
[   93.946637] NET: Registered protocol family 31
[   93.946645] Bluetooth: HCI device and connection manager initialized
[   93.946671] Bluetooth: HCI socket layer initialized
[   93.946687] Bluetooth: L2CAP socket layer initialized
[   93.946779] Bluetooth: SCO socket layer initialized
[   96.669415] fuse init (API version 7.26)
pi@raspberrypi:~ $ sudo modprobe esp8089
pi@raspberrypi:~ $
`

al177, 2018-02-05:
Er... I meant to say "lsmod", not "modprobe"

Anyway, it looks like it's working from your dmesg.  You won't see any output after "modprobe" unless there is a failure or if you're on a console receiving kernel log messages.

Run "ifconfig" and you should see the esp8266 on the wlan1 interface.

iHD992, 2018-02-05:
You are right. It works now. 
Maybe the Pi got confused about the two wifis (ESP8266 and USB-Wifi dongle) and was even more confused after I replaced the dongle by the keyboard. Also seems to need some time to connect after reboot.

closed 2018-02-05 by iHD992