al177/esp8089 #24: Driver crash under significant load

pkhanin, 2018-04-23:
Linux 4.14.34-v7+ #1110 SMP Mon Apr 16 15:18:51 BST 2018 armv7l GNU/Linux

               ***** EAGLE DRIVER VER:bdf5087c3deb*****

[    4.735612] ESP8089 reset via GPIO 0
[    4.953698] eagle_sdio_dummy: probe of mmc1:0001:1 failed with error -110
[    5.435826] mmc1: card 0001 removed
[    5.526507] mmc1: queuing unknown CIS tuple 0x01 (3 bytes)
[    5.533909] mmc1: queuing unknown CIS tuple 0x1a (5 bytes)
[    5.543321] mmc1: queuing unknown CIS tuple 0x1b (8 bytes)
[    5.545903] mmc1: queuing unknown CIS tuple 0x80 (1 bytes)
[    5.546101] mmc1: new high speed SDIO card at address 0001
[    5.546556] esp_sdio_dummy_probe enter
[    5.683737] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[    5.761396] esp_sdio_init power up OK
[    6.068760] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
[    6.307483] esp_host:bdf5087c3deb
               esp_target: e826c2b3c9fd 57 18202

[    6.307553] esp_readwrite_file: file /system/lib/modules/test_results filp_open error
[    6.307690] first normal exit
[    6.307909] esp_sdio_remove enter
[    6.421552] eagle_sdio: probe of mmc1:0001:1 failed with error -110
[    6.551788] mmc1: card 0001 removed
[    6.567778] random: crng init done
[    6.646136] sdio_read_cis: 2 callbacks suppressed
[    6.646144] mmc1: queuing unknown CIS tuple 0x01 (3 bytes)
[    6.653515] mmc1: queuing unknown CIS tuple 0x1a (5 bytes)
[    6.656730] mmc1: queuing unknown CIS tuple 0x1b (8 bytes)
[    6.659296] mmc1: queuing unknown CIS tuple 0x80 (1 bytes)
[    6.659378] mmc1: queuing unknown CIS tuple 0x81 (1 bytes)
[    6.659460] mmc1: queuing unknown CIS tuple 0x82 (1 bytes)
[    6.659507] mmc1: new high speed SDIO card at address 0001
[    7.055892] esp_host:bdf5087c3deb
               esp_target: e826c2b3c9fd 57 18202

[    7.413859] esp_op_add_interface STA
[    7.414016] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[    8.633558] wlan0: authenticate with fc:f5:28:4b:fa:f6
[    8.633774] wlan0: send auth to fc:f5:28:4b:fa:f6 (try 1/3)
[    8.640811] wlan0: authenticated
[    8.641429] wlan0: associate with fc:f5:28:4b:fa:f6 (try 1/3)
[    8.646420] wlan0: RX AssocResp from fc:f5:28:4b:fa:f6 (capab=0x11 status=0 aid=7)
[    8.646514] wlan0: associated
[    9.181937] esp_op_ampdu_action RX START fc:f5:28:4b:fa:f6 tid 0 3714
[    9.372877] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   11.738586] start tx ba session,addr:fc:f5:28:4b:fa:f6,tid:0
[   11.791395] esp_op_ampdu_action TX START, addr:fc:f5:28:4b:fa:f6,tid:0,state:1
[   11.793688] esp_op_ampdu_action TX OPERATION, addr:fc:f5:28:4b:fa:f6,tid:0,state:2
[   15.743887] esp_op_ampdu_action RX START fc:f5:28:4b:fa:f6 tid 6 44
[   17.389369] esp_op_ampdu_action RX START fc:f5:28:4b:fa:f6 tid 2 1
[   23.314979] start tx ba session,addr:fc:f5:28:4b:fa:f6,tid:5
[   23.351693] esp_op_ampdu_action TX START, addr:fc:f5:28:4b:fa:f6,tid:5,state:1
[   23.356201] esp_op_ampdu_action TX OPERATION, addr:fc:f5:28:4b:fa:f6,tid:5,state:2
[  142.003595] func sip_write_pkts err!!!!!!!!!: -84
[  142.003752] func sip_write_pkts err!!!!!!!!!: -110
[  142.003874] func sip_write_pkts err!!!!!!!!!: -110
[  142.003976] func sip_write_pkts err!!!!!!!!!: -110
[  142.004077] func sip_write_pkts err!!!!!!!!!: -110
[  142.004187] func sip_write_pkts err!!!!!!!!!: -110
[  142.004298] func sip_write_pkts err!!!!!!!!!: -110
[  142.310220] esp_sdio_remove enter
[  142.311254] sif_disable_irq release irq failed
[  142.419987] wlan0: deauthenticating from fc:f5:28:4b:fa:f6 by local choice (Reason: 3=DEAUTH_LEAVING)
[  142.481011] esp_op_ampdu_action RX STOP fc:f5:28:4b:fa:f6 tid 0
[  142.481028] esp_op_ampdu_action RX STOP fc:f5:28:4b:fa:f6 tid 2
[  142.481076] esp_op_ampdu_action RX STOP fc:f5:28:4b:fa:f6 tid 6
[  142.722277] mmc1: card 0001 removed
[  142.815686] mmc1: queuing unknown CIS tuple 0x01 (3 bytes)
[  142.823333] mmc1: queuing unknown CIS tuple 0x1a (5 bytes)
[... 41 lines cut ...]
[  142.975133] ff00: b6fd8958 80c02d00 00000008 b6fd8940 b698a198 b6fd8940 b6fd8958 80c02d00
[  142.979438] ff20: 00000008 b698a180 b6a3ff7c b6a3ff38 801377f0 80137344 b6a3ff5c b6a3e000
[  142.983741] ff40: 00000000 80c02d00 80c88122 b6a3e038 b698309c b6983080 00000000 b6983040
[  142.988117] ff60: b698a180 8013778c b698309c b6939e90 b6a3ffac b6a3ff80 8013d860 80137798
[  142.992512] ff80: 80102d94 b6983040 8013d724 00000000 00000000 00000000 00000000 00000000
[  142.996943] ffa0: 00000000 b6a3ffb0 8010810c 8013d730 00000000 00000000 00000000 00000000
[  143.001291] ffc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  143.005664] ffe0: 00000000 00000000 00000000 00000000 00000013 00000000 00000000 00000000
[  143.010043] [<7f2c4438>] (esp_sdio_probe [esp8089]) from [<80622f38>] (sdio_bus_probe+0x110/0x128)
[  143.014446] [<80622f38>] (sdio_bus_probe) from [<80542318>] (driver_probe_device+0x24c/0x308)
[  143.018859] [<80542318>] (driver_probe_device) from [<80542580>] (__device_attach_driver+0xa4/0xbc)
[  143.023355] [<80542580>] (__device_attach_driver) from [<80540400>] (bus_for_each_drv+0x74/0xa8)
[  143.027914] [<80540400>] (bus_for_each_drv) from [<80541f84>] (__device_attach+0xc0/0x120)
[  143.032464] [<80541f84>] (__device_attach) from [<805425b4>] (device_initial_probe+0x1c/0x20)
[  143.036983] [<805425b4>] (device_initial_probe) from [<80541474>] (bus_probe_device+0x94/0x9c)
[  143.041578] [<80541474>] (bus_probe_device) from [<8053f498>] (device_add+0x3f4/0x57c)
[  143.046190] [<8053f498>] (device_add) from [<806232d0>] (sdio_add_func+0x6c/0x80)
[  143.050789] [<806232d0>] (sdio_add_func) from [<80622520>] (mmc_attach_sdio+0x264/0x358)
[  143.055484] [<80622520>] (mmc_attach_sdio) from [<80619204>] (mmc_rescan+0x30c/0x394)
[  143.060210] [<80619204>] (mmc_rescan) from [<80137490>] (process_one_work+0x158/0x454)
[  143.064974] [<80137490>] (process_one_work) from [<801377f0>] (worker_thread+0x64/0x5b8)
[  143.069700] [<801377f0>] (worker_thread) from [<8013d860>] (kthread+0x13c/0x16c)
[  143.074491] [<8013d860>] (kthread) from [<8010810c>] (ret_from_fork+0x14/0x28)
[  143.079292] Code: e5954018 e3a01000 e5851018 e2863008 (e5948004)
[  143.084166] ---[ end trace 99c7c8ccce0684c9 ]---

al177, 2018-04-23:
What hardware are you using?

From the mmc driver messages it seems like the ESP took a reset and dropped off the SDIO bus.  This driver crashes if the device disappears from the bus before unloading.

al177, 2018-04-23:
10-15cm is too long for 50MHz SDIO without impedance controlled cabling.  You'll need to scope the signals at both ends of the bus to see if this is actually the issue.

Your error looks more like the ESP was reset unexpectedly.  The ESP8266 is power hungry.  If the power supply isn't solid at the input of the module, then you could get unexpected resets.  Try adding caps between power and ground as close as possible to the module (10-100uF) and scope it while under load.

pkhanin, 2018-04-23:
Okay, I will try to it. Glad to not hear, I found a problem in the driver. Thank you!

al177, 2018-04-23:
Keep this issue open until you confirm whether or not it's a hardware issue.

CRImier, 2018-05-01:
@pkhanin How are you loading the hardware? I have hardware that's known to be stable to me, and I can try doing the same thing you're doing.

closed 2018-07-05 by al177