al177/esp8089 #26: Linux driver esp8266 long ping

voloviq, 2018-06-05:
Hello,
I encounter some problem with access to the network through wifi. Device and driver connect to access point. The connection is stable. The problem is when I ping this device. Please take a look at the log

> PING 192.168.1.213 (192.168.1.213) 56(84) bytes of data.
64 bytes from 192.168.1.213: icmp_seq=1 ttl=64 time=802 ms
64 bytes from 192.168.1.213: icmp_seq=2 ttl=64 time=36.3 ms
64 bytes from 192.168.1.213: icmp_seq=3 ttl=64 time=753 ms
64 bytes from 192.168.1.213: icmp_seq=4 ttl=64 time=43.5 ms
64 bytes from 192.168.1.213: icmp_seq=5 ttl=64 time=193 ms
64 bytes from 192.168.1.213: icmp_seq=6 ttl=64 time=77.0 ms
64 bytes from 192.168.1.213: icmp_seq=7 ttl=64 time=56.0 ms
64 bytes from 192.168.1.213: icmp_seq=8 ttl=64 time=293 ms
64 bytes from 192.168.1.213: icmp_seq=9 ttl=64 time=2640 ms
64 bytes from 192.168.1.213: icmp_seq=10 ttl=64 time=1640 ms
64 bytes from 192.168.1.213: icmp_seq=11 ttl=64 time=640 ms
64 bytes from 192.168.1.213: icmp_seq=12 ttl=64 time=75.0 ms
64 bytes from 192.168.1.213: icmp_seq=13 ttl=64 time=240 ms
64 bytes from 192.168.1.213: icmp_seq=14 ttl=64 time=14.3 ms
64 bytes from 192.168.1.213: icmp_seq=15 ttl=64 time=3.10 ms
64 bytes from 192.168.1.213: icmp_seq=16 ttl=64 time=29.3 ms
64 bytes from 192.168.1.213: icmp_seq=17 ttl=64 time=114 ms
64 bytes from 192.168.1.213: icmp_seq=18 ttl=64 time=41.0 ms
64 bytes from 192.168.1.213: icmp_seq=19 ttl=64 time=207 ms
64 bytes from 192.168.1.213: icmp_seq=20 ttl=64 time=651 ms
64 bytes from 192.168.1.213: icmp_seq=21 ttl=64 time=1217 ms
64 bytes from 192.168.1.213: icmp_seq=22 ttl=64 time=639 ms
64 bytes from 192.168.1.213: icmp_seq=23 ttl=64 time=1252 ms
64 bytes from 192.168.1.213: icmp_seq=24 ttl=64 time=245 ms
64 bytes from 192.168.1.213: icmp_seq=25 ttl=64 time=548 ms
64 bytes from 192.168.1.213: icmp_seq=26 ttl=64 time=1122 ms
64 bytes from 192.168.1.213: icmp_seq=27 ttl=64 time=796 ms
64 bytes from 192.168.1.213: icmp_seq=28 ttl=64 time=365 ms
64 bytes from 192.168.1.213: icmp_seq=29 ttl=64 time=5048 ms
64 bytes from 192.168.1.213: icmp_seq=30 ttl=64 time=5027 ms
64 bytes from 192.168.1.213: icmp_seq=31 ttl=64 time=4027 ms
64 bytes from 192.168.1.213: icmp_seq=32 ttl=64 time=3027 ms
64 bytes from 192.168.1.213: icmp_seq=33 ttl=64 time=2027 ms
64 bytes from 192.168.1.213: icmp_seq=34 ttl=64 time=1027 ms
64 bytes from 192.168.1.213: icmp_seq=35 ttl=64 time=1931 ms
64 bytes from 192.168.1.213: icmp_seq=36 ttl=64 time=5387 ms
64 bytes from 192.168.1.213: icmp_seq=37 ttl=64 time=4387 ms
64 bytes from 192.168.1.213: icmp_seq=38 ttl=64 time=13706 ms
64 bytes from 192.168.1.213: icmp_seq=39 ttl=64 time=12698 ms
64 bytes from 192.168.1.213: icmp_seq=40 ttl=64 time=11690 ms
64 bytes from 192.168.1.213: icmp_seq=41 ttl=64 time=10682 ms
64 bytes from 192.168.1.213: icmp_seq=42 ttl=64 time=9681 ms
64 bytes from 192.168.1.213: icmp_seq=43 ttl=64 time=8674 ms
From 192.168.1.213 icmp_seq=88 Destination Host Unreachable
From 192.168.1.213 icmp_seq=89 Destination Host Unreachable
From 192.168.1.213 icmp_seq=90 Destination Host Unreachable
From 192.168.1.213 icmp_seq=91 Destination Host Unreachable
From 192.168.1.213 icmp_seq=92 Destination Host Unreachable
From 192.168.1.213 icmp_seq=93 Destination Host Unreachable
From 192.168.1.213 icmp_seq=94 Destination Host Unreachable
From 192.168.1.213 icmp_seq=95 Destination Host Unreachable
From 192.168.1.213 icmp_seq=96 Destination Host Unreachable
From 192.168.1.213 icmp_seq=97 Destination Host Unreachable
From 192.168.1.213 icmp_seq=98 Destination Host Unreachable
From 192.168.1.213 icmp_seq=99 Destination Host Unreachable
From 192.168.1.213 icmp_seq=100 Destination Host Unreachable
From 192.168.1.213 icmp_seq=101 Destination Host Unreachable
From 192.168.1.213 icmp_seq=102 Destination Host Unreachable
From 192.168.1.213 icmp_seq=103 Destination Host Unreachable
From 192.168.1.213 icmp_seq=104 Destination Host Unreachable
From 192.168.1.213 icmp_seq=105 Destination Host Unreachable
From 192.168.1.213 icmp_seq=106 Destination Host Unreachable
From 192.168.1.213 icmp_seq=107 Destination Host Unreachable
From 192.168.1.213 icmp_seq=108 Destination Host Unreachable
From 192.168.1.213 icmp_seq=109 Destination Host Unreachable
From 192.168.1.213 icmp_seq=110 Destination Host Unreachable
[... 88 lines cut ...]
Jun  4 14:00:05 v user.warn kernel: [    4.337886] esp_readwrite_file: file /system/lib/modules/test_results filp_open error
Jun  4 14:00:05 uuuu user.warn kernel: [    4.374375] esp_sdio_remove enter
Jun  4 14:00:05 uuuu user.warn kernel: [    5.000521] esp_sdio: initializing netlink
Jun  4 14:00:05 uuuu user.warn kernel: [    5.020455] esp_host:bdf5087c3deb
Jun  4 14:00:05 uuuu user.warn kernel: [    5.020455] esp_target: e826c2b3c9fd 57 18202
Jun  4 14:00:05 uuuu user.warn kernel: [   10.343256] esp_op_add_interface STA 
Jun  4 14:00:05 uuuu user.info kernel: [   11.728551] wlan0: RX AssocResp from b0:4e:26:f0:cf:a3 (capab=0x1431 status=0 aid=1)
Jun  4 14:00:05 uuuu user.warn kernel: [   11.743603] esp_op_ampdu_action RX START b0:4e:26:f0:cf:a3 tid 0 0
Jun  4 14:00:08 uuuu user.warn kernel: [   18.918399] esp_op_ampdu_action TX START, addr:b0:4e:26:f0:cf:a3,tid:0,state:1
Jun  4 14:00:08 uuuu user.warn kernel: [   18.948542] esp_op_ampdu_action TX OPERATION, addr:b0:4e:26:f0:cf:a3,tid:0,state:2

> cat /var/log/messages | grep wlan

> Jun  4 14:00:05 uuuu user.info kernel: [   10.347365] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Jun  4 14:00:05 uuuu user.info kernel: [   11.654622] wlan0: authenticate with b0:4e:26:f0:cf:a3
Jun  4 14:00:05 uuuu user.info kernel: [   11.660390] wlan0: send auth to b0:4e:26:f0:cf:a3 (try 1/3)
Jun  4 14:00:05 uuuu user.info kernel: [   11.678600] wlan0: authenticated
Jun  4 14:00:05 uuuu user.info kernel: [   11.688410] wlan0: associate with b0:4e:26:f0:cf:a3 (try 1/3)
Jun  4 14:00:05 uuuu user.info kernel: [   11.728551] wlan0: RX AssocResp from b0:4e:26:f0:cf:a3 (capab=0x1431 status=0 aid=1)
Jun  4 14:00:05 uuuu user.info kernel: [   11.736521] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
Jun  4 14:00:05 uuuu user.info kernel: [   11.743114] wlan0: associated_

And now is the best. I have some pice of hardware with the same firmware which works very well. I make changes hardware nothings helps.

I discovered also that when connect to old access point ping works normally. I'm confused what can be wrong, where to find a potential problem?

CRImier, 2018-06-05:
So, can the problem only be replicated on a specific piece of hardware?

voloviq, 2018-06-05:
Maybe it is good idea, I check my wifi dongle on RPI.

voloviq, 2018-06-13:
Hi again,
so on RPI, results are fantastic ping is 2ms sometimes 3ms transfer of file as well. Currently, I see that on my board still appears process from the driver. Process name is "195     2 root     SW       0   0%   1% [ksdioirqd/mmc0]". It does not take CPU to much but it constantly appears. This irq can be a cause of a problem?

al177, 2018-06-14:
It's normal.

SDIO doesn't generate hard interrupts on all host controller types, so this kthread listens to the bus and notifies the driver whenever there's a pending interrupt.

closed 2018-06-14 by al177