al177/esp8089 #28: Integration with kernel tree

pablogs9, 2018-11-15:
I'm trying to compile this module for a embedded device including it in our kernel tree branch. Does anyone knows how must be the Makefile rewriten to be included in Kbuild system and the .ko files generated for MODPOST?

CRImier, 2018-12-27:
@pablogs9 you will likely want to see this commit: https://github.com/Icenowy/esp8089/commit/5586b6aac8e7e7da4b0b8439e9c3876f11d8da90

zoobab, 2019-01-29:
What about submitting this linux driver upstream?

al177, 2019-01-31:
@zoobab I would love to see this mainlined, but I don't have any experience doing so.  Given the trouble others with more experience have had in getting this driver accepted, I don't think this is the driver to cut my upstream submission teeth on.

closed 2020-03-03 by pablogs9