al177/esp8089 #32: `dmks makedeb` fails on latest Debian (+ explanation)

CRImier, 2018-12-27:
Currently, [Debian Stretch (=> Ubuntu and Raspbian) has broken `dkms`](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=832558). So, when you'll try to run `dkms makedeb`, it will fail like this:

```
DKMS: mkdeb completed.
Moving built files to /var/lib/dkms/esp8089/1.9.20181227/deb...mv: cannot stat '/tmp/dkms.lal1zd/esp8089-dkms_1.9.20181227_armhf.deb': No such file or directory
(bad exit status: 1)
Cleaning up temporary files...
cp /var/lib/dkms/esp8089/1.9.20181227/deb/*.deb .
cp: cannot stat '/var/lib/dkms/esp8089/1.9.20181227/deb/*.deb': No such file or directory
Makefile:114: recipe for target 'dkmsdeb' failed
make: *** [dkmsdeb] Error 1
```

The solution, for now, is to patch `/usr/sbin/dkms` like this:

```
--- /usr/sbin/dkms      2018-12-27 18:04:49.045945342 +0200
+++ /usr/sbin/dkms.old  2018-12-27 17:58:01.039218120 +0200
@@ -3123,7 +3123,7 @@
             die 7 $"There was a problem creating your ${create_type}."
         echo $""
         echo $"DKMS: mk${create_type} completed."
-       invoke_command "mv '$temp_dir/${debian_package}-dkms_${module_version}_all.deb' '$deb_basedir'" "Moving built files to $deb_basedir"
+       invoke_command "mv '$temp_dir/${debian_package}-dkms_${module_version}_${debian_build_arch}.deb' '$deb_basedir'" "Moving built files to $deb_basedir"
        ;;
     bmdeb)
        export KVER="$kernelver"
```

Basically, replace the `${debian_build_arch}` by `all`. Attaching the .patch file in a .zip (because GitHub).
[dkms_fix.zip](https://github.com/al177/esp8089/files/2712842/dkms_fix.zip)

al177, 2018-12-27:
Thanks!  I'll review this as soon as I can

CRImier, 2018-12-27:
No need to review, just a tip in case you can't compile the .deb, too - which seems likely.

al177, 2018-12-27:
Sorry, I looked at this before I groked what was going on.  I have run into this before, and rediscover it every time I do a new release.  Thanks for the reminder!

al177, 2019-06-03:
Note that I have a workaround at the bottom of the README using the upstream DKMS instead of Debian's.

closed 2019-06-03 by al177