al177/esp8089 #33: How to configure AP mode?

haoning747, 2019-01-04:
Hello, I have completely bulid this driver and successfully use it in my board, but i don't know how to configure AP mode, can you give some direction? Eagerly  wait for your reply, thx.

haoning747, 2019-01-06:
Is anyone here? Ask for help....

al177, 2019-01-07:
Sorry for the delay, busy with work at the moment.

The driver supports AP mode, but you will need something like hostapd running to service the AP.

The [hostapd section of this guide](https://www.raspberrypi.org/documentation/configuration/wireless/access-point.md) is directly applicable to working with this driver.

haoning747, 2019-01-07:
> Sorry for the delay, busy with work at the moment.
>
> The driver supports AP mode, but you will need something like hostapd running to service the AP.
>
> The [hostapd section of this [...]

谢谢大神指点！万分感谢。

closed 2019-05-20 by al177