al177/esp8089 #35: Raspberry Pi Compute no wlan device added

mintu19, 2019-07-31:
I have successfully added esp-12f to pi compute on gpio 34-39 as SDIO with GPIO 0 as reset.
mmc1 detects the device. 

However ESP driver fails with this error on startup...
```
[   17.111769] resetting event timeout
[   17.111783] esp_init_all failed: -110
[   17.111788] first error exit
```

And when i try to reset gpio, 1) after using sudo modprobe -r esp8089, eso8089 is not removed (shows in lsmod)... but there is no error in output. Also there is no esp_host lines etc...

```
[  175.334908] esp_sdio_exit
[  175.335007] esp_sdio_remove enter
[  175.336774] ESP8089 reset via GPIO 0
[  175.915856] mmc1: card 0001 removed
[  175.968583] sdio_read_cis: 2 callbacks suppressed
[  175.968592] mmc1: queuing unknown CIS tuple 0x01 (3 bytes)
[  175.976021] mmc1: queuing unknown CIS tuple 0x1a (5 bytes)
[  175.979260] mmc1: queuing unknown CIS tuple 0x1b (8 bytes)
[  175.982159] mmc1: queuing unknown CIS tuple 0x80 (1 bytes)
[  175.982255] mmc1: queuing unknown CIS tuple 0x81 (1 bytes)
[  175.982352] mmc1: queuing unknown CIS tuple 0x82 (1 bytes)
[  175.982404] mmc1: new high speed SDIO card at address 0001
[  176.041852] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[  176.045385] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[  176.104264]
               ***** EAGLE DRIVER VER:bdf5087c3deb*****

[  176.104317] ESP8089 reset via GPIO 0
[  176.315543] eagle_sdio_dummy: probe of mmc1:0001:1 failed with error -110
[  177.035754] mmc1: card 0001 removed
[  177.087490] mmc1: queuing unknown CIS tuple 0x01 (3 bytes)
[  177.094894] mmc1: queuing unknown CIS tuple 0x1a (5 bytes)
[  177.098217] mmc1: queuing unknown CIS tuple 0x1b (8 bytes)
[  177.101131] mmc1: queuing unknown CIS tuple 0x80 (1 bytes)
[  177.101355] mmc1: new high speed SDIO card at address 0001
[  177.101809] esp_sdio_dummy_probe enter
[  177.315378] esp_sdio_init power up OK
```

This is before i can even reset gpio (if i try to reset gpio i get kernel panic).
After 10 sec of this i again get same message... 

```
[  177.101809] esp_sdio_dummy_probe enter
[  177.315378] esp_sdio_init power up OK
[  187.994606] resetting event timeout
[  187.994619] esp_init_all failed: -110
[  187.994625] first error exit
```

If i try to power down module by removing power, i get kernel panic.

I all of these cases, I am not getting wifi device shown as wlan0 or anything else.

mintu19, 2019-08-01:
ESP-12F output Serial

```
 ets 
 ets Jan  8 2013,rst cause:1, boot mode:(7,7)

waiting for host

 ets Jan  8 2013,rst cause:1, boot mode:(7,7)

waiting for host
ets Nov 21 2014 17:43:09
chip on
user code done
```

command output for reload: 

```
pi@raspberrypi:~/esp8089 $ sudo modprobe -r esp8089
pi@raspberrypi:~/esp8089 $ dmesg
[  826.500131] esp_sdio_exit
[  826.500227] esp_sdio_remove enter
[  826.505078] ESP8089 reset via GPIO 1
[  827.220586] mmc1: card 0001 removed
[  827.273143] mmc1: queuing unknown CIS tuple 0x01 (3 bytes)
[  827.280560] mmc1: queuing unknown CIS tuple 0x1a (5 bytes)
[  827.283788] mmc1: queuing unknown CIS tuple 0x1b (8 bytes)
[  827.286557] mmc1: queuing unknown CIS tuple 0x80 (1 bytes)
[  827.286647] mmc1: queuing unknown CIS tuple 0x81 (1 bytes)
[  827.286738] mmc1: queuing unknown CIS tuple 0x82 (1 bytes)
[  827.286788] mmc1: new high speed SDIO card at address 0001
[  827.346049] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[  827.349550] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[  827.400096]
               ***** EAGLE DRIVER VER:bdf5087c3deb*****

[  827.400158] ESP8089 reset via GPIO 1
[  827.610450] eagle_sdio_dummy: probe of mmc1:0001:1 failed with error -110
[  828.340743] mmc1: card 0001 removed
[  828.392491] mmc1: queuing unknown CIS tuple 0x01 (3 bytes)
[  828.399899] mmc1: queuing unknown CIS tuple 0x1a (5 bytes)
[  828.403151] mmc1: queuing unknown CIS tuple 0x1b (8 bytes)
[  828.405927] mmc1: queuing unknown CIS tuple 0x80 (1 bytes)
[  828.406139] mmc1: new high speed SDIO card at address 0001
[  828.406584] esp_sdio_dummy_probe enter
[  828.620418] esp_sdio_init power up OK
[  838.900735] resetting event timeout
[  838.900747] esp_init_all failed: -110
[  838.900753] first error exit
pi@raspberrypi:~/esp8089 $
```

MMC info:

```
pi@raspberrypi:~/esp8089 $ sudo cat /sys/kernel/debug/mmc1/ios
clock:          50000000 Hz
actual clock:   41666667 Hz
vdd:            21 (3.3 ~ 3.4 V)
bus mode:       2 (push-pull)
chip select:    0 (don't care)
power mode:     2 (on)
bus width:      2 (4 bits)
timing spec:    2 (sd high-speed)
signal voltage: 0 (3.30 V)
driver type:    0 (driver type B)
```

I am using 33 ohms in series with sdio lines and using short distance to cmio board.
Using 3.3k pullup on esp pin 3 (CH_PD). and also 10uF across power lines.

mintu19, 2019-08-03:
Changing the module fixed the issue. Old module was not working with this. 
But the same module worked with AT commands but not with drivers.

CRImier, 2019-08-03:
How did you change it? Can you share your changes? I.e. `git diff`?

mintu19, 2019-08-03:
I changed my module... hardware. ESP-12f.

CRImier, 2019-08-03:
Ah, got it. I also have encountered incompatibility issues with certain ESP-12F issues - limited to certain makes of the module, it seems. What kind of kernel panics do you get and when?

Also, just a tip - `dmesg -w &` is a good command to use while debugging, runs as a background job of the shell and shows you the `dmesg` output as it is created, giving you quicker feedback about your actions.

mintu19, 2019-08-03:
It was my CH_PD GPIO hanging. I was using external pullup (3.3 k) as i read somewhere and chip was not working. And even touching CH_PD pin gave kernal panic (with driver i guess). And restarting driver did not detect card. I got cannot reset GPIO on pin 5 error. I removed external pullup. Wifi is working now properly.
Only problem is that I have to manually restart the driver after reboot or else keep getting mmc1 errors of card not detected. Tried writing startup script to hold CH_PD low on start for half a sec to reset. But doesn't work. Have to manually start after all reboot. It works when i halt and cut power but without power cycle, have to do it manually.

mintu19, 2019-08-04:
Any way to restart driver/device on reboot?

al177, 2019-08-04:
Like the USB subsystem, the SDIO driver will probe for new devices and load the appropriate module if available. If you install the module with "sudo make install" or through the DKMS package then the module should be registered to be loaded when the driver sees the ESP on the bus.

I have also noticed that some ESP-12F modules, even from the same vendor, have issues with CH_PD not having a sufficient internal pull. It's been a while, but I think I used 10k pullups successfully.

mintu19, 2019-08-05:
I have used 3.3K external pullup but it was being problematic. I will try with bigger 10k to see if anything happens.

If i halt the system and do power cycle then drivers are detected (without external pullup). But if I reboot, driver doesn't load. I have to manually load via modprobe. I get this....

```
[    6.083146] mmc1: Controller never released inhibit bit(s).
[    6.083177] mmc1: Got command interrupt 0x00030000 even though no command operation was in progress.
[    6.133560] mmc-bcm2835 3f300000.mmc: no support for card's volts
[    6.133580] mmc1: error -22 whilst initialising SDIO card
[    6.143843] mmc1: Controller never released inhibit bit(s).
[    6.143867] mmc1: Got command interrupt 0x00030000 even though no command operation was in progress.
[    6.536030] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[    6.570899] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[    6.664034] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
[   20.183234] mmc-bcm2835 3f300000.mmc: no support for card's volts
[   20.183248] mmc1: error -22 whilst initialising SDIO card
[   20.193507] mmc1: Controller never released inhibit bit(s).
[   20.193531] mmc1: Got command interrupt 0x00030000 even though no command operation was in progress.
[   22.712776] mmc1: Controller never released inhibit bit(s).
[   22.712801] mmc1: Got command interrupt 0x00030000 even though no command operation was in progress.
[   22.762663] mmc-bcm2835 3f300000.mmc: no support for card's volts
[   22.762674] mmc1: error -22 whilst initialising SDIO card
[   22.772934] mmc1: Controller never released inhibit bit(s).
[   22.772956] mmc1: Got command interrupt 0x00030000 even though no command operation was in progress.
[   26.482633] mmc1: Got command interrupt 0x00030000 even though no command operation was in progress.
[   26.482648] mmc1: Got command interrupt 0x00030000 even though no command operation was in progress.
[   26.492966] mmc1: Controller never released inhibit bit(s).
[   26.523794] mmc1: Got command interrupt 0x00030000 even though no command operation was in progress.
[   26.523808] mmc1: Got command interrupt 0x00030000 even though no command operation was in progress.
[   26.535373] mmc1: Controller never released inhibit bit(s).
[   26.542256] mmc-bcm2835 3f300000.mmc: no support for card's volts
[   26.542267] mmc1: error -22 whilst initialising SDIO card
.
.
.
.
```

Also on reboot, the blue led on esp-12f doesn't blink.

mintu19, 2019-08-28:
Any update to reset on reboot?

apica-alfred, 2020-06-18:
I'm facing the same issue here. It seems to properly work after powercycling the board (10K pullup). Also, I'm directly using ESP8266EX chip. @mintu19 did you managed to workaround this?

apica-alfred, 2020-06-18:
In fact, I'm not really sure I have the same issue... there's an `file /system/lib/modules/test_results filp_open error` before the -110

```
Jun 18 12:45:41 pi kernel: [    3.944678] ***** EAGLE DRIVER VER:bdf5087c3deb*****
Jun 18 12:45:41 pi kernel: [    3.944678]
Jun 18 12:45:41 pi kernel: [    3.944759] ESP8089 reset via GPIO 6
Jun 18 12:45:41 pi kernel: [    3.992270] mmc1: card 0001 removed
Jun 18 12:45:41 pi kernel: [    4.272045] mmc1: queuing unknown CIS tuple 0x01 (3 bytes)
Jun 18 12:45:41 pi kernel: [    4.290461] mmc1: queuing unknown CIS tuple 0x1a (5 bytes)
Jun 18 12:45:41 pi kernel: [    4.296730] mmc1: queuing unknown CIS tuple 0x1b (8 bytes)
Jun 18 12:45:41 pi kernel: [    4.302069] mmc1: queuing unknown CIS tuple 0x80 (1 bytes)
Jun 18 12:45:41 pi kernel: [    4.302411] mmc1: new high speed SDIO card at address 0001
Jun 18 12:45:41 pi kernel: [    4.302995] esp_sdio_dummy_probe enter
Jun 18 12:45:41 pi kernel: [    4.521477] esp_sdio_init power up OK
Jun 18 12:45:41 pi kernel: [    5.068460] esp_host:bdf5087c3deb
Jun 18 12:45:41 pi kernel: [    5.068460] esp_target: e826c2b3c9fd 57 18202
Jun 18 12:45:41 pi kernel: [    5.068460]
Jun 18 12:45:41 pi kernel: [    5.068519] esp_readwrite_file: file /system/lib/modules/test_results filp_open error
Jun 18 12:45:41 pi kernel: [    5.068743] first normal exit
Jun 18 12:45:41 pi kernel: [    5.068948] esp_sdio_remove enter
Jun 18 12:45:41 pi kernel: [    5.181643] eagle_sdio: probe of mmc1:0001:1 failed with error -110
Jun 18 12:45:41 pi kernel: [    5.352281] mmc1: card 0001 removed
Jun 18 12:45:41 pi kernel: [    5.468633] mmc1: new high speed SDIO card at address 0001
Jun 18 12:45:42 pi kernel: [    5.840689] esp_host:bdf5087c3deb
Jun 18 12:45:42 pi kernel: [    5.840689] esp_target: e826c2b3c9fd 57 18202
Jun 18 12:45:42 pi kernel: [    5.840689]
Jun 18 12:45:42 pi kernel: [    6.021045] esp_op_add_interface STA
Jun 18 12:45:42 pi kernel: [    6.021197] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Jun 18 12:45:42 pi dhcpcd[378]: wlan0: waiting for carrier
Jun 18 12:45:42 pi dhcpcd[378]: wlan0: carrier acquired
Jun 18 12:45:42 pi dhcpcd[378]: wlan0: carrier lost
Jun 18 12:45:47 pi dhcpcd[378]: wwan0: DHCP lease expired
Jun 18 12:45:47 pi dhcpcd[378]: wwan0: soliciting a DHCP lease
Jun 18 12:46:06 pi dhcpcd[378]: wwan0: carrier lost
Jun 18 12:46:06 pi avahi-daemon[371]: Leaving mDNS multicast group on interface wwan0.IPv6 with address fe80::80e4:ebff:fef3:8b33.
Jun 18 12:46:06 pi avahi-daemon[371]: Withdrawing address record for fe80::80e4:ebff:fef3:8b33 on wwan0.
Jun 18 12:46:06 pi dhcpcd[378]: wwan0: removing interface
Jun 18 12:46:11 pi dhcpcd[378]: timed out
Jun 18 12:46:11 pi dhcpcd[378]: forked to background, child pid 541
Jun 18 12:51:17 pi kernel: [  321.705287] esp_op_add_interface STA
Jun 18 12:51:17 pi kernel: [  321.705480] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Jun 18 12:56:16 pi kernel: [  621.504421] esp_op_add_interface STA
Jun 18 12:56:16 pi kernel: [  621.504636] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
```

apica-alfred, 2020-06-18:
🤔 also I'm seeing that `iwlist wlan0 scan` correctly reports some nets. Wpa supplicant configuration is right though (If I connect a wifi dongle, it connects instantly).

```
Jun 18 18:46:17 pi kernel: [ 3138.719263] esp_op_add_interface STA
Jun 18 18:46:17 pi kernel: [ 3138.719996] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Jun 18 18:46:43 pi kernel: [ 3165.645982] wlan0: authenticate with 84:aa:9c:a1:a5:64
Jun 18 18:46:43 pi kernel: [ 3165.646080] wlan0: send auth to 84:aa:9c:a1:a5:64 (try 1/3)
Jun 18 18:46:44 pi kernel: [ 3165.854580] wlan0: send auth to 84:aa:9c:a1:a5:64 (try 2/3)
Jun 18 18:46:44 pi kernel: [ 3166.064594] wlan0: send auth to 84:aa:9c:a1:a5:64 (try 3/3)
Jun 18 18:46:44 pi kernel: [ 3166.274587] wlan0: authentication with 84:aa:9c:a1:a5:64 timed out
```

I'm using Raspbian 4.14.79-v7+ on a CM3 core. @al177 any clue?

Thank you!

al177, 2022-08-15:
Is the dongle 5GHz band? I have a problem using ESPs at work since our guest network is 5GHz only, so dual band devices work while ESPs don't. Since you're using a "down" ESP with your own antenna and coupling, it's possible that a weak connection to the AP would manifest like this. What does the signal strength of your target SSID look like during the scan?

Sorry this got stale. Closing this for now, but if you want to continue please reopen this issue.

closed 2022-08-15 by al177