al177/esp8089 #36: Driver not working in Raspi 4 Raspbian Buster

thecalle, 2019-08-01:
Insmod gives some errors:

[  959.705105] esp8089: Unknown symbol ieee80211_stop_tx_ba_session (err -2)
[  959.705196] esp8089: Unknown symbol ieee80211_start_tx_ba_cb_irqsafe (err -2)
[  959.705384] esp8089: Unknown symbol ieee80211_free_hw (err -2)
[  959.705457] esp8089: Unknown symbol ieee80211_start_tx_ba_session (err -2)
[  959.705515] esp8089: Unknown symbol ieee80211_register_hw (err -2)
[  959.705613] esp8089: Unknown symbol ieee80211_get_buffered_bc (err -2)
[  959.705746] esp8089: Unknown symbol ieee80211_stop_tx_ba_cb_irqsafe (err -2)
[  959.705866] esp8089: Unknown symbol ieee80211_remain_on_channel_expired (err -2)
[  959.705935] esp8089: Unknown symbol ieee80211_wake_queues (err -2)
[  959.705998] esp8089: Unknown symbol ieee80211_alloc_hw_nm (err -2)
[  959.706074] esp8089: Unknown symbol ieee80211_tx_status (err -2)
[  959.706122] esp8089: Unknown symbol ieee80211_ready_on_channel (err -2)
[  959.706163] esp8089: Unknown symbol ieee80211_stop_queues (err -2)
[  959.706292] esp8089: Unknown symbol ieee80211_scan_completed (err -2)
[  959.706369] esp8089: Unknown symbol ieee80211_unregister_hw (err -2)
[  959.706424] esp8089: Unknown symbol ieee80211_beacon_get_tim (err -2)
[  959.706530] esp8089: Unknown symbol ieee80211_queue_work (err -2)
[  959.706611] esp8089: Unknown symbol ieee80211_rx_napi (err -2)

al177, 2019-08-03:
I haven't tested against the new Buster distro. Are you trying to use this with a Pi 4?

al177, 2019-08-04:
I tested the last release (1.9.20190603) against Buster 2019-07-10 and was able to get it to work. The only catch is Buster seems to need rfkill disabled before the interface can be used, which I don't think is specific to the esp8089.

From the symbol errors, I'd guess that mac80211 or cfg80211 modules are not installed.

closed 2019-08-04 by al177