al177/esp8089 #42: Firmware failed to download

duanearnett, 2020-04-23:
Running `sudo rmmod esp8089` and I get two flashes on the **ESP-12f** and the following in `dmesg`:

```
[  249.133658] esp_sdio_exit 
[  249.133770] esp_sdio_remove enter
[  249.133793] sip_detach wrong state 1
[  249.139089] ESP8089 reset via GPIO 0
[  250.083981] mmc1: card 0001 removed
[  250.136872] sdio_read_cis: 2 callbacks suppressed
[  250.136882] mmc1: queuing unknown CIS tuple 0x01 (3 bytes)
[  250.144570] mmc1: queuing unknown CIS tuple 0x1a (5 bytes)
[  250.147817] mmc1: queuing unknown CIS tuple 0x1b (8 bytes)
[  250.150653] mmc1: queuing unknown CIS tuple 0x80 (1 bytes)
[  250.150747] mmc1: queuing unknown CIS tuple 0x81 (1 bytes)
[  250.150838] mmc1: queuing unknown CIS tuple 0x82 (1 bytes)
[  250.150889] mmc1: new high speed SDIO card at address 0001
[  250.169807] 
               ***** EAGLE DRIVER VER:bdf5087c3deb*****

[  250.169952] ESP8089 reset via GPIO 0
[  250.383786] eagle_sdio_dummy: probe of mmc1:0001:1 failed with error -110
[  251.203994] mmc1: card 0001 removed
[  251.256849] mmc1: queuing unknown CIS tuple 0x01 (3 bytes)
[  251.264598] mmc1: queuing unknown CIS tuple 0x1a (5 bytes)
[  251.267827] mmc1: queuing unknown CIS tuple 0x1b (8 bytes)
[  251.270675] mmc1: queuing unknown CIS tuple 0x80 (1 bytes)
[  251.270890] mmc1: new high speed SDIO card at address 0001
[  251.271390] esp_sdio_dummy_probe enter
[  251.483596] esp_sdio_init power up OK
[  251.514337] sip_write_memory send buffer failed
[  251.514352] esp_download_fw Failed to write fw, err: -84
[  251.514358] download firmware failed
[  251.514364] esp_init_all failed: -84
[  251.514369] first error exit
```

This is on a new RPI CM3+ module via headers on an IO Board. Any troubleshooting steps on why the firmware fails to load? I will try another module later but don't want to bin this one if its a software issue.

al177, 2020-04-27:
I would check signal integrity on the SDIO signals and the reset pin. Then I'd check that the power supply is solid to the ESP module. It can pull significant current on startup.

Then make sure it's a 'real' ESP-12F and not a knockoff with different flash configuration or different mapping of the flash castellations.

duanearnett, 2020-05-05:
I did shorten the jumpers to the ESP-12F but it did not improve. I'm going to put the chip on a better protoboard and hope that improves the signal integrity. Thank you for the response!

closed 2020-05-07 by al177

duanearnett, 2020-05-20:
New chip (from Waveshare) and new proto board. `dmesg` gives me the following: 

```
esp_sdio_init power up OK
  sip_rx first read err -84 12
resetting event timeout
esp_init_all failed -110
first error exit
```

Trying to listen to the serial interface but just getting gibberish when I reload the driver, no matter the baudrate I select. This is on a waveshare compute module board with bank 0 GPIO which I confirmed raspi-gpio was reporting the right functionality for the GPIO. Using 33ohm resistors on the transmission lines but not pulling anything else down or up? I've come across varying sources saying they needed pull-ups on reset or IO15. Once schematic I found had 200ohm resistors for the transmission lines? Not sure if this helps someone with more experience pinpoint my issue, but thanks for the help!

duanearnett, 2021-03-19:
Wanted to follow-up here. I flashed an official AI Thinker ESP-12F with 2.2.0 of the firmware. I have this chip on a milled version of a prototype board with 33ohm resistors in series for all the data lines. I can get serial feedback only if I pull GPIO0 low and reset, but all AT commands work, I get an AP, etc. Its on a compute module 3 IO board, powered off a bench power supply.

```
cat /sys/kernel/debug/mmc1/ios                                                                                                                                                                                                                                                

clock:          50000000 Hz
actual clock:   41666667 Hz
vdd:            21 (3.3 ~ 3.4 V)
bus mode:       2 (push-pull)
chip select:    0 (don't care)
power mode:     2 (on)
bus width:      2 (4 bits)
timing spec:    2 (sd high-speed)
signal voltage: 0 (3.30 V)
driver type:    0 (driver type B)
```

dmesg output: 
```
               ***** EAGLE DRIVER VER:bdf5087c3deb*****

[  611.146382] ESP8089 reset via GPIO 5
[  611.358980] eagle_sdio_dummy: probe of mmc1:0001:1 failed with error -110
[  611.519262] mmc1: card 0001 removed
[  611.573829] mmc1: queuing unknown CIS tuple 0x01 (3 bytes)
[  611.581769] mmc1: queuing unknown CIS tuple 0x1a (5 bytes)
[  611.585261] mmc1: queuing unknown CIS tuple 0x1b (8 bytes)
[  611.589312] mmc1: queuing unknown CIS tuple 0x80 (1 bytes)
[  611.589518] mmc1: queuing unknown CIS tuple 0x81 (1 bytes)
[  611.589656] mmc1: queuing unknown CIS tuple 0x82 (1 bytes)
[  611.589731] mmc1: new high speed SDIO card at address 0001
[  611.590272] esp_sdio_dummy_probe enter
[  611.808758] esp_sdio_init power up OK
[  622.558809] resetting event timeout
[  622.558826] esp_init_all failed: -110
[  622.558836] first error exit
```

Any other tricks to this? Thanks for the help!

duanearnett, 2021-03-21:
Wanted to do another follow-up and say this was most likely because of the CMIO board I was using. A custom printed module, with the same wiring runs perfectly. Thanks @al177 for all your work and helping through everyone's issues.