al177/esp8089 #43: Any reason not to work on kernel 5.4.51 ?

amateusz, 2020-08-19:
I compiled the module on current stable (5.4.51 as of July 2020) and it doesn't seem to work. I confirmed it working on 4.9 some time ago on the same HW (RPi A+ with soldered ESP-12F).

Could you confirm it's not working on kernels ≥5 ?

---

By the way thank you for your work. I've spotted your pants project on hackaday some long time ago

al177, 2020-09-09:
I just tried a fresh install on a Pi 2B of Raspbian Lite and the release 1.9.20190603 deb and it appears to be working on the latest kernel 5.4.51-v7+.

However I haven't tried on an armv6 platform. Can you elaborate on what's not working?

amateusz, 2020-09-17:
I confirm it works again on 5.4.51+ armv6.

The problem was missing `dtoverlay=sdio,poll_once=off` in `config.txt` on a fresh install. Found it in [your hackaday instructions](https://hackaday.io/project/8678-rpi-wifi). 
Maybe consider adding it to readme?

Sorry for the fuss.

---

**"Retro" RPi 1A going strong:**
[image]

al177, 2020-11-09:
Thanks for letting me know, and glad you got it working!

closed 2020-11-09 by al177