al177/esp8089 #44: esp8089 on kernel 5.9-rc6

valadaa48, 2020-09-30:
Hello, I have this driver working on kernel 4.4 but when trying to bring up the driver (compiled in the kernel) on 5.9 I get this:

```
               ***** EAGLE DRIVER VER:bdf5087c3deb*****

[    3.006872] ESP8089 reset via GPIO 105
[    3.013548] esp_sdio_dummy_probe enter
[    3.222248] esp_sdio_init power up OK
[    3.685645] esp_host:bdf5087c3deb
               esp_target: e826c2b3c9fd 57 18202

[    3.701133] esp_readwrite_file: file /system/lib/modules/test_results filp_open error
[    3.713367] first normal exit
[    3.718507] esp_sdio_remove enter
[    3.723892] sif_disable_irq release irq failed
```

I assume the `sif_disable_irq release irq failed` entry indicates an error? That line isn't present in kernel 4.4. Any idea where I should start looking?

valadaa48, 2020-09-30:
I also see

```
esp_opp_add_interface STA
esp_opp_add_interface only support MAX 2 interface
```
being printed over and over to my tty after bootup

al177, 2022-08-15:
I did some cleanup for kernels 5.4.x and later since you asked this. If you haven't tried already, give the latest a shot.

closed 2022-08-15 by al177