al177/esp8089 #45: esp8089 on licheepi_nano with 4.15 kernel

masoudr, 2021-01-14:
Hi,
I tried to compile and install the driver on my board but I get the following output in kernel log (I'm not sure if it is an error):
```
Starting mdev... OK
[   17.143437] esp8089: loading out-of-tree module taints kernel.
[   17.158502] 
[   17.158502] ***** EAGLE DRIVER VER:bdf5087c3deb*****
[   17.158502] 
[   17.166587] ESP8089 reset via GPIO 0
[   17.388537] esp_sdio_dummy_probe enter
[   17.608205] esp_sdio_init power up OK
[   18.092809] esp_host:bdf5087c3deb
[   18.092809] esp_target: e826c2b3c9fd 57 18202
[   18.092809] 
[   18.102158] esp_readwrite_file: file /system/lib/modules/test_results filp_open error
[   18.115300] first normal exit
[   18.118864] esp_sdio_remove enter
[   18.125392] sif_disable_irq release irq failed
[   18.248642] eagle_sdio: probe of mmc0:0001:1 failed with error -110
Saving random seed: OK

Welcome 
F1C100S login: [   18.968783] mmc0: card 0001 removed
[   19.063403] mmc0: queuing unknown CIS tuple 0x01 (3 bytes)
[   19.076367] mmc0: queuing unknown CIS tuple 0x1a (5 bytes)
[   19.086700] mmc0: queuing unknown CIS tuple 0x1b (8 bytes)
[   19.094962] mmc0: queuing unknown CIS tuple 0x80 (1 bytes)
[   19.102165] mmc0: queuing unknown CIS tuple 0x81 (1 bytes)
[   19.107783] mmc0: queuing unknown CIS tuple 0x82 (1 bytes)
[   19.114742] mmc0: new high speed SDIO card at address 0001
[   19.548256] esp_host:bdf5087c3deb
[   19.548256] esp_target: e826c2b3c9fd 57 18202
[   19.548256]
```
mmc0 on dts:
```
...
&mmc0 {
	vmmc-supply = <&reg_vcc3v3>;
	bus-width = <4>;
	broken-cd;
	status = "okay";
};
...
```
The chip is connected within the memory card slot.
I can see the wlan0 with `ip a` command but I can't connect to any network with `wpa_supplicant`, I get the below error (I'm not sure if this is related to driver or not):

```bash
# wpa_supplicant -B -i wlan0 -c /etc/wpa_supplicant.conf -D wext
Successfully initialized wpa_supplicant
rfkill: Cannot o[   73.555268] esp_op_add_interface STA 
pen RFKILL control device
ioctl[SIOCSIWENCODEEXT]: Invalid argument
ioctl[SIOCSIWENCODEEXT]: Invalid argument 
```
Thanks.

al177, 2021-01-18:
I haven't tried this driver with any other platforms than Pi Zero / A+ / 2B+ so I can't help much with platform specifics. I know that @Icenowy has a branch for LicheePi plus their custom SDIO card in this fork: https://github.com/Icenowy/esp8089/tree/cleanup . 

It's been a while so I could be mistaken, but I thought it was normal to see some probe errors. Out of reset the ESP is running a bootstrap program out of ROM that establishes SDIO comms with the host and waits for the runtime firmware to be downloaded into RAM. Once the firmware is downloaded and starts executing, the ESP has to redo the SDIO handshake which produces that error.

As for the ioctl error, that may have to do with wpa_supplicant [(first Google hit here)](https://askubuntu.com/questions/652864/wpa-supplicant-does-not-connect-with-ioctlsiocsiwencodeext-invalid-argument), but I wonder if the RFKILL error may be related.

Feel free to respond to this, but as I'm certain it's not a driver error I'm going to mark this closed.

closed 2021-01-18 by al177

masoudr, 2021-01-19:
Thanks for the info, I tried the driver you mentioned but get the exact same behavior. The `/etc/network/interfaces` didn't work for me as it can't connect to my AP. I tried to use wpa_supplicant and then try the wpa_cli, but whenever I scan for APs I get only one or two APs and I even can't see my own AP in the scan result list. So I can't connect to either network. I'm also using an external antenna on the board. After one or two minutes of scan I get some error messages in the console.
I'm adding the full console log maybe it helps.

```
Starting kernel ...                                                                                                                
                                                                                                                                   
[    0.000000] Booting Linux on physical CPU 0x0                                                                                   
[    0.000000] Linux version 5.2.0-licheepi-nano (masoud@ubuntu) (gcc version 7.5.0 (Linaro GCC 7.5-2019.12)) #3 Mon Jan 18 00:09:1
[    0.000000] CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=0005317f                                                       
[    0.000000] CPU: VIVT data cache, VIVT instruction cache                                                                        
[    0.000000] OF: fdt: Machine model: Lichee Pi Nano                                                                              
[    0.000000] Memory policy: Data cache writeback                                                                                 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 8128                                                         
[    0.000000] Kernel command line: console=ttyS0,115200 panic=5 rootwait root=/dev/mtdblock3 rw rootfstype=jffs2                  
[    0.000000] Dentry cache hash table entries: 4096 (order: 2, 16384 bytes)                                                       
[    0.000000] Inode-cache hash table entries: 2048 (order: 1, 8192 bytes)                                                         
[    0.000000] Memory: 22716K/32768K available (6144K kernel code, 221K rwdata, 1480K rodata, 1024K init, 230K bss, 10052K reserve)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1                                                          
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16                                                                     
[    0.000000] random: get_random_bytes called from start_kernel+0x254/0x3cc with crng_init=0                                      
[    0.000055] sched_clock: 32 bits at 24MHz, resolution 41ns, wraps every 89478484971ns                                           
[    0.000141] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 79635851949 ns                            
[    0.000716] Console: colour dummy device 80x30                                                                                  
[    0.000817] Calibrating delay loop... 203.16 BogoMIPS (lpj=1015808)                                                             
[    0.070263] pid_max: default: 32768 minimum: 301                                                                                
[    0.070720] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)                                                         
[    0.070772] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)                                                    
[    0.072574] CPU: Testing write buffer coherency: ok                                                                             
[    0.074982] Setting up static identity map for 0x80100000 - 0x80100058                                                          
[    0.077409] devtmpfs: initialized                                                                                               
[    0.082962] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns                    
[    0.083027] futex hash table entries: 256 (order: -1, 3072 bytes)                                                               
[    0.083368] pinctrl core: initialized pinctrl subsystem                                                                         
[    0.086078] NET: Registered protocol family 16                                                                                  
[    0.088095] DMA: preallocated 256 KiB pool for atomic coherent allocations                                                      
[    0.090503] cpuidle: using governor menu                                                                                        
[    0.143380] SCSI subsystem initialized                                                                                          
[    0.143810] usbcore: registered new interface driver usbfs                                                                      
[    0.143981] usbcore: registered new interface driver hub                                                                        
[    0.144189] usbcore: registered new device driver usb                                                                           
[    0.144679] pps_core: LinuxPPS API ver. 1 registered                                                                            
[... cut ...]
[  564.417023] 1fa0:                                     00000000 00000000 00000000 00000000                                       
[  564.425190] 1fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000                                       
[  564.433351] 1fe0: 00000000 00000000 00000000 00000000 00000013 00000000                                                         
[  564.439966] Code: e8bd8070 e3510000 0a00000d e3a03000 (e4d0e001)                                                                
[  564.446050] ---[ end trace d2b818adfb1568b7 ]---
```