al177/esp8089 #46: esp8089 sdio + imx6ull yocto Linux 4.19.35

wKolesnikAndrew, 2021-01-20:
Hello, I am try to run esp8089 chip with Linux 4.19.35 (yocto project on imx6ull controller).

My device tree looks:

&usdhc1 {
	pinctrl-names = "default";
	pinctrl-0 = <&pinctrl_usdhc1>;
        card-detect-delay = <800>;
        status = "okay";
        wakeup-source;

	esp8089: sdio_wifi@1 {
                vmmc-supply = <&reg_sd1_vmmc>;
		compatible = "esp,esp8089";
		esp,crystal_26M_en = <2>;
		clock-frequency = <50000000>;
		clock-freq-min-max = <200000 50000000>;
		supports-highspeed;
		supports-sdio;
		bus-width = <4>;
		cd-post;
		pm-ignore-notify;
		wifi-host; 
		non-removable;
		disable-wp;
                status = "okay";
	};
};

CH_PD connected to GPIO 9 imx6ull, and I am fix schematic to pullup this pin.
After powerup mmc0 device recognized like high speed SDIO card: 

# cat /sys/kernel/debug/mmc0/ios
clock:          50000000 Hz
actual clock:   44000000 Hz
vdd:            21 (3.3 ~ 3.4 V)
bus mode:       2 (push-pull)
chip select:    0 (don't care)
power mode:     2 (on)
bus width:      2 (4 bits)
timing spec:    2 (sd high-speed)
signal voltage: 0 (3.30 V)
driver type:    0 (driver type B)

[    0.991350] sdhci: Secure Digital Host Controller Interface driver
[    0.991370] sdhci: Copyright(c) Pierre Ossman
[    0.991381] sdhci-pltfm: SDHCI platform and OF driver helper
[    1.028674] mmc0: SDHCI controller on 2190000.usdhc [2190000.usdhc] using ADMA
[    1.055568] mmc0: queuing unknown CIS tuple 0x01 (3 bytes)
[    1.066636] mmc0: queuing unknown CIS tuple 0x1a (5 bytes)
[    1.070629] mmc0: queuing unknown CIS tuple 0x1b (8 bytes)
[    1.075118] mmc1: SDHCI controller on 2194000.usdhc [2194000.usdhc] using ADMA
[    1.088194] random: fast init done
[    1.089401] mmc0: queuing unknown CIS tuple 0x80 (1 bytes)
[    1.089635] mmc0: queuing unknown CIS tuple 0x81 (1 bytes)
[    1.089889] mmc0: queuing unknown CIS tuple 0x82 (1 bytes)
[    1.090013] mmc0: new high speed SDIO card at address 0001

But wheh driver tries go up i have an error:

[   13.123295] esp8089: loading out-of-tree module taints kernel.
[   13.207291] imx6ull_reset_off EXT RESET NOW OFF!
[   13.207349]
               ***** EAGLE DRIVER VER:bdf5087c3deb*****

[   13.207474] ESP8089 reset via GPIO 9
[   13.465308] sdio_register_driver 1
[   13.474356] eagle_sdio_dummy: probe of mmc0:0001:1 failed with error -110

[   24.805251] esp_sdio_init ------ RETRY ------
[   24.808095] ESP8089 reset via GPIO 9
[   25.065279] sdio_register_driver 1
[   25.074315] eagle_sdio_dummy: probe of mmc0:0001:1 failed with error -110
[   36.325233] esp_sdio_init ------ RETRY ------
[   36.325429] ESP8089 reset via GPIO 9
[   36.585211] sdio_register_driver 1
[   36.594276] eagle_sdio_dummy: probe of mmc0:0001:1 failed with error -110
[   47.845224] esp_sdio_init ------ RETRY ------
[   47.845423] ESP8089 reset via GPIO 9
[   48.105166] sdio_register_driver 1
[   48.114775] eagle_sdio_dummy: probe of mmc0:0001:1 failed with error -110
[   59.365185] esp_sdio_init ------ RETRY ------
[   59.365381] eagle sdio can not power up!

So, I can see than sdio device not reconnected. Maybe system configurated to scaning one time peripheral devices.
Can you help me to run esp8089 driver?

Schematic and dmesg in attached.
[dmesg.txt](https://github.com/al177/esp8089/files/5841250/dmesg.txt)
[image]
[image]

al177, 2021-01-20:
Hi wKolesnikAndrew,

I'm not sure if I can fully help you with debugging this, but you may want to look at an earlier design I did with a bare ESP8266 and it looks similar. The exceptions are EXT_RES is floating / no-connect in my design and I have 33 ohm series resistors for all of the SDIO signals. The project is here if you want to compare notes: https://github.com/al177/esp_hat

One thing I found is that CH_PD / CHIP_EN needs to be driven low for reset BUT should not be driven when released. I can't remember what the issue was though.

wKolesnikAndrew, 2021-01-21:
Hello! Thanks for your answer.

1. CH_PD is driven for reset and reconfigureted to input when released.
2. Made the EXT_RES floating (I disconnected it from the circuit).
**3. When starting the driver, manually disconnect and reconnect the sdio-device:**

 echo -n "2190000.usdhc" > /sys/devices/soc0/soc/2100000.aips-bus/2190000.usdhc/driver/unbind (disconnect sdio)
echo -n "2190000.usdhc" > /sys/bus/soc/devices/soc0/soc/2100000.aips-bus/2190000.usdhc/subsystem/drivers/sdhci-esdhc-imx/bind (connect sdio)
**and after this dummy driver go up:**

              ***** EAGLE DRIVER VER:bdf5087c3deb*****

[  +0.000101] ESP8089 reset via GPIO 9
[  +0.214795] sdio_register_driver 1
[  +0.009174] eagle_sdio_dummy: probe of mmc0:0001:1 failed with error -110
[  +2.081379] mmc0: card 0001 removed                                                                                        **(<----  disconnected mmc0)**
[  +3.739124] mmc0: SDHCI controller on 2190000.usdhc [2190000.usdhc] using ADMA
[  +0.035496] mmc0: queuing unknown CIS tuple 0x01 (3 bytes)
[  +0.013725] mmc0: queuing unknown CIS tuple 0x1a (5 bytes)
[  +0.009254] mmc0: queuing unknown CIS tuple 0x1b (8 bytes)
[  +0.036516] mmc0: queuing unknown CIS tuple 0x80 (1 bytes)
[  +0.001922] mmc0: queuing unknown CIS tuple 0x81 (1 bytes)
[  +0.000678] mmc0: queuing unknown CIS tuple 0x82 (1 bytes)
[  +0.000683] mmc0: new high speed SDIO card at address 0001                                                      **(<----  connected mmc0)**
[  +0.013637] esp_sdio_dummy_probe enter
[  +0.218336] esp_sdio_init power up OK                                                  **(<----  dummy driver OK?)**    
[  +0.000301] sdio_register_driver 1
[  +0.644199] esp_host:bdf5087c3deb
              esp_target: e826c2b3c9fd 57 18202

****, but another error appier:**** 

[  +0.011556] Unable to handle kernel NULL pointer dereference at virtual addres                                 s 00000000
[  +0.000030] pgd = df9d2411
[  +0.000011] [00000000] *pgd=00000000
[  +0.000029] Internal error: Oops: 80000007 [#1] PREEMPT SMP ARM
[  +0.000010] Modules linked in: esp8089(O+)
[  +0.000033] CPU: 0 PID: 80 Comm: kworker/u2:1 Tainted: G           O      4.19                                 .35-1.1.0+g0f9917c56d59 #1
[  +0.000010] Hardware name: Freescale i.MX6 UltraLite (Device Tree)
[  +0.000131] Workqueue: esp_wkq sip_rxq_process [esp8089]
[  +0.000019] PC is at   (null)
[  +0.000078] LR is at esp_readwrite_file+0x108/0x238 [esp8089]
[  +0.000012] pc : [<00000000>]    lr : [<7f04af3c>]    psr: a00e0013
[  +0.000011] sp : 8852baf0  ip : 000f2174  fp : 88a9f040
[  +0.000010] r10: 887eb7e4  r9 : 00000000  r8 : 00000038
[  +0.000011] r7 : 8852bb24  r6 : 00000000  r5 : 8852bc24  r4 : 887c0e40
[  +0.000011] r3 : 887c0e80  r2 : 00000038  r1 : 8852bc24  r0 : 887c0e40
[  +0.000015] Flags: NzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment none
[  +0.000012] Control: 10c53c7d  Table: 8902406a  DAC: 00000051
[  +0.000013] Process kworker/u2:1 (pid: 80, stack limit = 0x6387d74a)
[  +0.000013] Stack: (0x8852baf0 to 0x8852c000)
[  +0.000017] bae0:                                     88a9f040 80a5fbdc 8852bb                                 10 81004d08
[  +0.000021] bb00: 8852bb24 887eb780 7f056080 88f3a180 7f056080 7f04f228 000000                                 16 88a9f04e
[  +0.000021] bb20: 81004d08 7379732f 2f6d6574 2f62696c 75646f6d 2f73656c 747365                                 74 7365725f
[  +0.000020] bb40: 73746c75 8015e000 00000000 00000001 00000000 8852bbe0 8832a4                                 80 8bb3c200
[  +0.000020] bb60: 81004d70 8bb3c180 8832a400 8bb3c180 8bb3c180 8832a400 8bb3c1                                 80 00000000
[  +0.000020] bb80: 8bb3c200 8015966c 8bb3c180 8832a400 8852bbe4 80a6679c 8bb3c1                                 80 80150ce0
[  +0.000020] bba0: 8bb3c290 00000000 ffffffff 00000000 8895a000 8014d054 8832a4                                 00 00000001
[  +0.000020] bbc0: 8895a000 8bb3c180 8832a400 882de400 81004d08 89052e00 000000                                 00 80b02b18
[  +0.000021] bbe0: 8852bc2c 80a61b28 00000000 8852bc30 7fffffff 8852bd44 8852bd                                 48 80a61f00
[  +0.000020] bc00: 81004d08 8b0c90b5 00000000 ffffe000 8852bd44 8852bd48 81004d                                 08 81004d08
[  +0.000020] bc20: 00000000 5f707365 74736f68 6664623a 37383035 65643363 73650a                                 62 61745f70
[  +0.000021] bc40: 74656772 3865203a 32633632 39633362 35206466 38312037 0a3230                                 32 8920de00
[  +0.000020] bc60: 8920df40 8bb3c180 8bb3c340 80a666f4 8832a400 00000000 81004d                                 08 81004e98
[  +0.000021] bc80: 8bb3c180 8015162c 8832a400 8bb3c180 8852bca4 80151ce4 8920de                                 00 8bb3c180
[  +0.000020] bca0: 8852bcd4 80151d04 81004e98 00000000 8920e270 8920de00 8bb3c1                                 80 81004d08
[  +0.000020] bcc0: 81004e98 00000000 8852bd0c 80a66748 8920de00 80152f7c 0abb90                                 00 60030193
[  +0.000020] bce0: 00000000 8b0c90b5 00000000 88983db8 00000001 88983dc4 000000                                 00 8015318c
[  +0.000020] bd00: 81004d08 00000000 00000001 88d50b40 00000001 8074b3dc 88d50b                                 c0 00000000
[  +0.000020] bd20: 00000030 8074a3a8 88d50b40 00000001 00000001 80a666f4 88d50b                                 40 8074835c
[... 29 lines cut ...]
[... cut ...]
imx6ull14x14evk kernel: bae0:                                     88a9f040 80a5f                                 bdc 8852bb10 81004d08
[  +0.000016] bfe0: 00000000 00000000 00000000 00000000 00000013 00000000
[  +0.000018] Code: bad PC value
[  +0.001057] ---[ end trace 373e885131263860 ]---
........
[Apr23 10:53] resetting event timeout
[  +0.000031] esp_init_all failed: -110
[  +0.000039] first error exit

What could be the problem?
Full dmesg in attach.
[dmesg.txt](https://github.com/al177/esp8089/files/5848477/dmesg.txt)

wKolesnikAndrew, 2021-01-25:
Hello, issue can be closed.
I am added flag ESP_ACK_INTERRUPT and rebuild driver.
Additional in file sdio_stub.h I have replaced strange function extern void sdmmc_ack_interrupt(struct mmc_host *mmc); to dummy:
void sdmmc_ack_interrupt(struct mmc_host *mmc) { return; }.
And after that driver run normally. Thanks.

[   15.977106] imx6ull_reset_off EXT RESET NOW OFF!
[   15.977307]
               ***** EAGLE DRIVER VER:bdf5087c3deb*****

[   15.977480] ESP8089 reset via GPIO 9
[   16.195073] sdio_register_driver 1
[   16.204038] eagle_sdio_dummy: probe of mmc0:0001:1 failed with error -110
[   16.234879] Generic PHY 2188000.ethernet-1:00: attached PHY driver [Generic PHY] (mii_bus:phy_addr=2188000.ethernet-1:00, irq=POLL)
[   16.235857] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   18.325730] fec 2188000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
[   18.325792] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   19.425408] Generic PHY 2188000.ethernet-1:00: attached PHY driver [Generic PHY] (mii_bus:phy_addr=2188000.ethernet-1:00, irq=POLL)
[   19.427316] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   21.525652] fec 2188000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
[   21.525715] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   23.569703] FAT-fs (sda1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[   27.362182] EXT4-fs (mmcblk1p2): mounted filesystem with ordered data mode. Opts: (null)
[   27.366061] esp_sdio_init ------ RETRY ------
[   27.366263] ESP8089 reset via GPIO 9
[   27.585164] ESP8089 reset via GPIO 9
[   27.805063] sdio_register_driver 1
[   27.814086] eagle_sdio_dummy: probe of mmc0:0001:1 failed with error -110
[   38.885034] esp_sdio_init ------ RETRY ------
[   38.885279] ESP8089 reset via GPIO 9
[   39.105123] ESP8089 reset via GPIO 9
[   39.325115] sdio_register_driver 1
[   39.333911] eagle_sdio_dummy: probe of mmc0:0001:1 failed with error -110
[   50.405097] esp_sdio_init ------ RETRY ------
[   50.405281] ESP8089 reset via GPIO 9
[   50.625046] ESP8089 reset via GPIO 9
[   50.845311] sdio_register_driver 1
[   50.854160] eagle_sdio_dummy: probe of mmc0:0001:1 failed with error -110
[   61.925207] esp_sdio_init ------ RETRY ------
[   61.925388] ESP8089 reset via GPIO 9
[   62.145138] eagle sdio can not power up!
[  118.174240] imx6ull_reset_off EXT RESET NOW OFF!
[  118.174303]
               ***** EAGLE DRIVER VER:bdf5087c3deb*****

[  118.174397] ESP8089 reset via GPIO 9
[  118.385149] sdio_register_driver 1
[  118.393769] eagle_sdio_dummy: probe of mmc0:0001:1 failed with error -110
[  121.678621] mmc0: card 0001 removed
[  126.525123] mmc0: SDHCI controller on 2190000.usdhc [2190000.usdhc] using ADMA
[  126.558444] mmc0: queuing unknown CIS tuple 0x01 (3 bytes)
[  126.573369] mmc0: queuing unknown CIS tuple 0x1a (5 bytes)
[  126.579419] mmc0: queuing unknown CIS tuple 0x1b (8 bytes)
[  126.604176] mmc0: queuing unknown CIS tuple 0x80 (1 bytes)
[  126.607484] mmc0: queuing unknown CIS tuple 0x81 (1 bytes)
[  126.607685] mmc0: queuing unknown CIS tuple 0x82 (1 bytes)
[  126.607784] mmc0: new high speed SDIO card at address 0001
[  126.619158] esp_sdio_dummy_probe enter
[  126.834977] esp_sdio_init power up OK
[  126.835271] sdio_register_driver 1
[  127.418603] esp_host:bdf5087c3deb
               esp_target: e826c2b3c9fd 57 18202

[  127.418740] esp_readwrite_file: file /system/lib/modules/test_results filp_open error
[  127.418908] first normal exit
[  127.419233] esp_sdio_remove enter
[  127.424291] sif_disable_irq release irq failed
[  127.534990] sdio_register_driver 1
[  127.543784] eagle_sdio: probe of mmc0:0001:1 failed with error -110
[  129.362583] mmc0: card 0001 removed
[  131.807183] mmc0: SDHCI controller on 2190000.usdhc [2190000.usdhc] using ADMA
[  131.842261] mmc0: queuing unknown CIS tuple 0x01 (3 bytes)
[  131.856435] mmc0: queuing unknown CIS tuple 0x1a (5 bytes)
[  131.862950] mmc0: queuing unknown CIS tuple 0x1b (8 bytes)
[  131.876506] mmc0: queuing unknown CIS tuple 0x80 (1 bytes)
[  131.876679] mmc0: queuing unknown CIS tuple 0x81 (1 bytes)
[  131.876840] mmc0: queuing unknown CIS tuple 0x82 (1 bytes)
[  131.876934] mmc0: new high speed SDIO card at address 0001
[  132.363195] esp_host:bdf5087c3deb
               esp_target: e826c2b3c9fd 57 18202

[  132.651912] esp_op_add_interface STA
[  132.667367] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  172.687963] esp_op_change_interface enter,change to AP
[  172.687991] esp_op_change_interface type from 2 to 3
[  177.828714] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready

closed 2021-01-25 by wKolesnikAndrew