al177/esp8089 #47: Compile issue with rpi kernel 5.10.11-v7+

hackintoshlover, 2021-02-06:


hackintoshlover, 2021-02-06:
pi@raspberrypi:~/esp8089 $ sudo make
make -C /usr/src/linux-headers-5.10.11-v7+ M=/home/pi/esp8089
make[1]: Entering directory '/usr/src/linux-headers-5.10.11-v7+'
  CC [M]  /home/pi/esp8089/esp_debug.o
/home/pi/esp8089/esp_debug.c: In function ‘esp_dump_var’:
/home/pi/esp8089/esp_debug.c:90:20: error: void value not ignored as it ought to be
                 rc = debugfs_create_u8(name, mode, parent, (u8*)value);
                    ^
/home/pi/esp8089/esp_debug.c:93:20: error: void value not ignored as it ought to be
                 rc = debugfs_create_u16(name, mode, parent, (u16*)value);
                    ^
/home/pi/esp8089/esp_debug.c:96:20: error: void value not ignored as it ought to be
                 rc = debugfs_create_u32(name, mode, parent, (u32*)value);
                    ^
/home/pi/esp8089/esp_debug.c:99:20: error: void value not ignored as it ought to be
                 rc = debugfs_create_u64(name, mode, parent, (u64*)value);
                    ^
/home/pi/esp8089/esp_debug.c:109:20: error: void value not ignored as it ought to be
                 rc = debugfs_create_u32(name, mode, parent, (u32*)value);
                    ^
make[2]: *** [scripts/Makefile.build:279: /home/pi/esp8089/esp_debug.o] Error 1
make[1]: *** [Makefile:1808: /home/pi/esp8089] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.10.11-v7+'
make: *** [Makefile:126: modules] Error 2

duanearnett, 2021-03-24:
@hackintoshlover Removing `-DDEBUG_FS` from line 64 of the Makefile works for me to get it to compile on 5.10. I'm sure this is not the best fix, but it works if you need it.

SNO95, 2021-05-16:
Removing the "rc =" from the case structure in esp_debug.c also gets the driver to compile, but I don't know either if this is a good workaround. [](https://stackoverflow.com/a/17915388) here Jamin says that:
> This is a GCC error message that means the return-value of a function is 'void', but that you are trying to assign it to a non-void variable.

So maybe rc is a variable and can't have the value of a function?

On a side note I'm having trouble with the gpio reset, the driver can't operate the gpio pin but I'm able to with: 

`echo 43 (custom board, I went with this pin) >> /sys/class/gpio/export`
`echo high >> /sys/class/gpio/gpio43/direction`

I have to do this twice as the driver turns the pin off and removes the gpio43 fiolder from /sys/class/gpio, but after this everything works. I am not sure if this is because of the new kernel or something else, the compiled version worked flawlessly for me before. I am also not sure if I should open a new issue because of this.

al177, 2021-06-26:
5.4.0rc1 introduced a patch that makes all single value debugfs_create_* functions return void instead of the dentry. The only consumer in this driver doesn't care about the return anyway. Likely we don't need so many debug flags turned on but I'll leave that to users to decide.

I just committed patches for this and some long-standing warnings.

closed 2021-06-26 by al177