al177/esp8089 #51: Kernel 5.3.5 compilation error

YusufYYK, 2022-08-02:
Hi,

When i compile module with kernel 5.3.5 this error occurs on stage 2. How can i fix it? In this web page these errors given as warning. (https://whycan.com/t_4326.html) 

ubuntu@ubuntu:~/Desktop/pi/esp8089$ make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- -C ../linux-5.3.5/ M=$(pwd) modules
make: Entering directory '/home/ubuntu/Desktop/pi/linux-5.3.5'
  CC [M]  /home/ubuntu/Desktop/pi/esp8089/esp_debug.o
  CC [M]  /home/ubuntu/Desktop/pi/esp8089/sdio_sif_esp.o
  CC [M]  /home/ubuntu/Desktop/pi/esp8089/spi_sif_esp.o
  CC [M]  /home/ubuntu/Desktop/pi/esp8089/esp_io.o
  CC [M]  /home/ubuntu/Desktop/pi/esp8089/esp_file.o
  CC [M]  /home/ubuntu/Desktop/pi/esp8089/esp_main.o
  CC [M]  /home/ubuntu/Desktop/pi/esp8089/esp_sip.o
  CC [M]  /home/ubuntu/Desktop/pi/esp8089/esp_ext.o
  CC [M]  /home/ubuntu/Desktop/pi/esp8089/esp_ctrl.o
  CC [M]  /home/ubuntu/Desktop/pi/esp8089/esp_mac80211.o
  CC [M]  /home/ubuntu/Desktop/pi/esp8089/esp_utils.o
  CC [M]  /home/ubuntu/Desktop/pi/esp8089/esp_pm.o
  CC [M]  /home/ubuntu/Desktop/pi/esp8089/testmode.o
  LD [M]  /home/ubuntu/Desktop/pi/esp8089/esp8089.o
  Building modules, stage 2.
  MODPOST 1 modules
ERROR: "ieee80211_rx_napi" [/home/ubuntu/Desktop/pi/esp8089/esp8089.ko] undefined!
ERROR: "ieee80211_queue_work" [/home/ubuntu/Desktop/pi/esp8089/esp8089.ko] undefined!
ERROR: "ieee80211_hdrlen" [/home/ubuntu/Desktop/pi/esp8089/esp8089.ko] undefined!
ERROR: "ieee80211_beacon_get_tim" [/home/ubuntu/Desktop/pi/esp8089/esp8089.ko] undefined!
ERROR: "ieee80211_unregister_hw" [/home/ubuntu/Desktop/pi/esp8089/esp8089.ko] undefined!
ERROR: "ieee80211_scan_completed" [/home/ubuntu/Desktop/pi/esp8089/esp8089.ko] undefined!
ERROR: "wiphy_rfkill_start_polling" [/home/ubuntu/Desktop/pi/esp8089/esp8089.ko] undefined!
ERROR: "ieee80211_stop_queues" [/home/ubuntu/Desktop/pi/esp8089/esp8089.ko] undefined!
ERROR: "ieee80211_ready_on_channel" [/home/ubuntu/Desktop/pi/esp8089/esp8089.ko] undefined!
ERROR: "ieee80211_tx_status" [/home/ubuntu/Desktop/pi/esp8089/esp8089.ko] undefined!
ERROR: "ieee80211_alloc_hw_nm" [/home/ubuntu/Desktop/pi/esp8089/esp8089.ko] undefined!
ERROR: "ieee80211_wake_queues" [/home/ubuntu/Desktop/pi/esp8089/esp8089.ko] undefined!
ERROR: "ieee80211_remain_on_channel_expired" [/home/ubuntu/Desktop/pi/esp8089/esp8089.ko] undefined!
ERROR: "ieee80211_stop_tx_ba_cb_irqsafe" [/home/ubuntu/Desktop/pi/esp8089/esp8089.ko] undefined!
ERROR: "wiphy_rfkill_set_hw_state" [/home/ubuntu/Desktop/pi/esp8089/esp8089.ko] undefined!
ERROR: "ieee80211_get_buffered_bc" [/home/ubuntu/Desktop/pi/esp8089/esp8089.ko] undefined!
ERROR: "ieee80211_register_hw" [/home/ubuntu/Desktop/pi/esp8089/esp8089.ko] undefined!
ERROR: "ieee80211_start_tx_ba_session" [/home/ubuntu/Desktop/pi/esp8089/esp8089.ko] undefined!
ERROR: "ieee80211_free_hw" [/home/ubuntu/Desktop/pi/esp8089/esp8089.ko] undefined!
ERROR: "ieee80211_start_tx_ba_cb_irqsafe" [/home/ubuntu/Desktop/pi/esp8089/esp8089.ko] undefined!
ERROR: "ieee80211_stop_tx_ba_session" [/home/ubuntu/Desktop/pi/esp8089/esp8089.ko] undefined!
make[1]: *** [scripts/Makefile.modpost:103: modules-modpost] Error 1
make: *** [Makefile:1630: modules] Error 2
make: Leaving directory '/home/ubuntu/Desktop/pi/linux-5.3.5'

Thanks,

al177, 2022-08-06:
Make sure you have CONFIG_CFG80211 and CONFIG_MAC80211 in the kernel config.

YusufYYK, 2022-08-09:
Hi,

After kernel rebuild(5.4.209 with your instruction) the device works. I can connect to network and ping the router. But after some time it always occurs a problem and wlan0 is disappeared. Until the restart again module work. How can i fix it? 

64 bytes from 192.168.88.1: icmp_seq=289 ttl=64 time=5.43 ms
64 bytes from 192.168.88.1: icmp_seq=290 ttl=64 time=5.52 ms
64 bytes from 192.168.88.1: icmp_seq=291 ttl=64 time=6.89 ms
64 bytes from 192.168.88.1: icmp_seq=292 ttl=64 time=5.45 ms
64 bytes from 192.168.88.1: icmp_seq=293 ttl=64 time=5.84 ms
64 bytes from 192.168.88.1: icmp_seq=294 ttl=64 time=5.38 ms
64 bytes from 192.168.88.1: icmp_seq=295 ttl=64 time=6.21 ms
64 bytes from 192.168.88.1: icmp_seq=296 ttl=64 time=5.43 ms
64 bytes from 192.168.88.1: icmp_seq=297 ttl=64 time=1.78 ms
64 bytes from 192.168.88.1: icmp_seq=298 ttl=64 time=5.41 ms
64 bytes from 192.168.88.1: icmp_seq=299 ttl=64 time=5.35 ms
64 bytes from 192.168.88.1: icmp_seq=300 ttl=64 time=4.91 ms
64 bytes from 192.168.88.1: icmp_seq=301 ttl=64 time=5.99 ms
64 bytes from 192.168.88.1: icmp_seq=302 ttl=64 time=1.95 ms
64 bytes from 192.168.88.1: icmp_seq=303 ttl=64 time=5.36 ms
64 bytes from 192.168.88.1: icmp_seq=304 ttl=64 time=95.4 ms

---------- dmesg----------

[  +1,320559] wlan0: authenticate with 98:0d:67:15:6d:47
[  +0,005315] wlan0: send auth to 98:0d:67:15:6d:47 (try 1/3)
[  +0,010860] wlan0: authenticated
[  +0,012267] wlan0: associate with 98:0d:67:15:6d:47 (try 1/3)
[  +0,011682] wlan0: RX AssocResp from 98:0d:67:15:6d:47 (capab=0x431 status=0 aid=3)
[  +0,007939] wlan0: associated
[  597.844199] esp_sdio_remove enter
[  597.848717] sif_disable_irq release irq failed
[  597.853185] ------------[ cut here ]------------
[  597.857929] WARNING: CPU: 0 PID: 360 at kernel/workqueue.c:3049 __flush_work+0xac/0xb8
[  597.865884] Modules linked in: esp8089(O)
[  597.869911] CPU: 0 PID: 360 Comm: kworker/0:0 Tainted: G           O      5.4.209 #1
[  597.877651] Hardware name: Allwinner sun8i Family
[  597.882365] Workqueue: events_freezable mmc_rescan
[  597.887182] [<c010e510>] (unwind_backtrace) from [<c010b04c>] (show_stack+0x10/0x14)
[  597.894934] [<c010b04c>] (show_stack) from [<c07aa2e8>] (dump_stack+0x8c/0xa0)
[  597.902165] [<c07aa2e8>] (dump_stack) from [<c07a4388>] (__warn+0xc4/0xe0)
[  597.909045] [<c07a4388>] (__warn) from [<c07a441c>] (warn_slowpath_fmt+0x78/0xac)
[  597.916532] [<c07a441c>] (warn_slowpath_fmt) from [<c0139128>] (__flush_work+0xac/0xb8)
[  597.924539] [<c0139128>] (__flush_work) from [<c0139720>] (__cancel_work_timer+0x19c/0x238)
[  597.932930] [<c0139720>] (__cancel_work_timer) from [<bf0054c4>] (sip_detach+0x88/0x1c4 [esp8089])
[  597.941919] [<bf0054c4>] (sip_detach [esp8089]) from [<bf000e50>] (esp_sdio_remove+0x64/0x1a4 [esp8089])
[  597.951416] [<bf000e50>] (esp_sdio_remove [esp8089]) from [<c05a1338>] (sdio_bus_remove+0x30/0x11c)
[  597.960471] [<c05a1338>] (sdio_bus_remove) from [<c049dba4>] (__device_release_driver+0x148/0x204)
[  597.969435] [<c049dba4>] (__device_release_driver) from [<c049dc80>] (device_release_driver+0x20/0x2c)
[  597.978744] [<c049dc80>] (device_release_driver) from [<c049cf3c>] (bus_remove_device+0xcc/0xf8)
[  597.987532] [<c049cf3c>] (bus_remove_device) from [<c0498fc0>] (device_del+0x13c/0x450)
[  597.995538] [<c0498fc0>] (device_del) from [<c05a1548>] (sdio_remove_func+0x1c/0x28)
[  598.003283] [<c05a1548>] (sdio_remove_func) from [<c05a0450>] (mmc_sdio_detect+0x94/0x144)
[  598.011547] [<c05a0450>] (mmc_sdio_detect) from [<c0596fc0>] (mmc_rescan+0xcc/0x430)
[  598.019294] [<c0596fc0>] (mmc_rescan) from [<c01379ec>] (process_one_work+0x1d4/0x434)
[  598.027214] [<c01379ec>] (process_one_work) from [<c0137cb0>] (worker_thread+0x64/0x5b0)
[  598.035309] [<c0137cb0>] (worker_thread) from [<c013da94>] (kthread+0x14c/0x150)
[  598.042701] [<c013da94>] (kthread) from [<c01010e8>] (ret_from_fork+0x14/0x2c)
[  598.049923] Exception stack(0xc3847fb0 to 0xc3847ff8)
[  598.054976] 7fa0:                                     00000000 00000000 00000000 00000000
[  598.063154] 7fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  598.071330] 7fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[  598.078032] ---[ end trace aa0009ebe579ec9e ]---
[  598.082704] wlan0: deauthenticating from 98:0d:67:15:6d:47 by local choice (Reason: 3=DEAUTH_LEAVING)
[Oca 1 00:09] esp_sdio_remove enter
[  +0,004518] sif_disable_irq release irq failed
[  +0,004468] ------------[ cut here ]------------
[  +0,004744] WARNING: CPU: 0 PID: 360 at kernel/workqueue.c:3049 __flush_work+0xac/0xb8
[  +0,007955] Modules linked in: esp8089(O)
[  +0,004027] CPU: 0 PID: 360 Comm: kworker/0:0 Tainted: G           O      5.4.209 #1
[  +[  598.195164] mmc1: card 0001 removed
[... 8 lines cut ...]
[  +0,008391] [<c0139720>] (__cancel_work_timer) from [<bf0054c4>] (sip_detach+0x88/0x1c4 [esp8089])
[[  598.276647] sdio_read_cis: 2 callbacks suppressed
  +0,008989] [<bf0054c4>] (sip_detach [esp8089]) from [<[  598.276659] mmc1: queuing unknown CIS tuple 0x01 (3 bytes)
bf000e50>] (esp_sdio_remove+0x64/0x1a4 [esp8089])
[  +0,009497] [<bf000e50>] (esp_sdio_remove [esp8089]) from [<c05a1338>] (sdio_bus_remove+0x30/0x11c)
[  +0,009055] [<c05a1338>] (sdio_bus_remove) from [<c049dba4>] (__device_release_driver+0x148/0x204)
[  +0,008964] [<c049dba4>] (__device_release_driver) from [<c049dc80>] (device_release_driver+0x20/0x2c)
[... cut ...]
[  +0,005053] 7fa0:                                     00000000 00000000 00000000 00000000
[  +0,008178] 7fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  +0,008176] 7fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[  +0,006702] ---[ end trace aa0009ebe579ec9e ]---
[  +0,004672] wlan0: deauthenticating from 98:0d:67:15:6d:47 by local choice (Reason: 3=DEAUTH_LEAVING)
[  +0,112460] mmc1: card 0001 removed
[  +0,081483] sdio_read_cis: 2 callbacks suppressed
[  +0,000012] mmc1: queuing unknown CIS tuple 0x01 (3 bytes)

YusufYYK, 2022-08-11:
Hi,

I am using breadboard and esp8266-201(removed spi flash) kit with allwinner v3s(licheepi zero board). VCC pins connected to upper power line  and GNDs are  connected to  bottom line of breadboard. After using osciloscope, i see that when the wpa_supplicant connect to network,  the power line pk to pk voltage drops reach 800 mV and very noisy. I connect Vcc and Gnd closely on same line and add bypass caps. And add pull up 10k to SDIO pins , 330R series resistors. I checked whole circuit with osciloscope, every thing is good. I use trigger function to catch the reset pin but when error occurs, nothing change on reset pin. I thing it is a soft issue. Something cause to remove sdio. Just insmod and rmmod result is below paste.Same result with rmmod. What is your opinion? How can i fix it?

When error occurs:
-----------------------------
64 bytes from 192.168.88.1: icmp_seq=871 ttl=64 time=2.49 ms
64 bytes from 192.168.88.1: icmp_seq=872 ttl=64 time=7.16 ms
64 bytes from 192.168.88.1: icmp_seq=873 ttl=64 time=1.68 ms
[  985.843772] esp_sdio_remove enter
[  985.848314] sif_disable_irq release irq failed
[  985.852780] ------------[ cut here ]------------
[  985.857515] WARNING: CPU: 0 PID: 337 at kernel/workqueue.c:3049 __flush_work+0xac/0xb8
[  985.865467] Modules linked in: esp8089(O)
[  985.869495] CPU: 0 PID: 337 Comm: kworker/0:1 Tainted: G           O      5.4.209 #1
[  985.877235] Hardware name: Allwinner sun8i Family
[  985.881951] Workqueue: events_freezable mmc_rescan
[  985.886763] [<c010e510>] (unwind_backtrace) from [<c010b04c>] (show_stack+0x10/0x14)
[  985.894516] [<c010b04c>] (show_stack) from [<c07a83a0>] (dump_stack+0x8c/0xa0)
[  985.901748] [<c07a83a0>] (dump_stack) from [<c07a2468>] (__warn+0xc4/0xe0)
[  985.908627] [<c07a2468>] (__warn) from [<c07a24fc>] (warn_slowpath_fmt+0x78/0xac)
[  985.916107] [<c07a24fc>] (warn_slowpath_fmt) from [<c0139128>] (__flush_work+0xac/0xb8)
[  985.924115] [<c0139128>] (__flush_work) from [<c0139720>] (__cancel_work_timer+0x19c/0x238)
[  985.932505] [<c0139720>] (__cancel_work_timer) from [<bf0054c4>] (sip_detach+0x88/0x1c4 [esp8089])
[  985.941497] [<bf0054c4>] (sip_detach [esp8089]) from [<bf000e50>] (esp_sdio_remove+0x64/0x1a4 [esp8089])
[  985.950994] [<bf000e50>] (esp_sdio_remove [esp8089]) from [<c05a1338>] (sdio_bus_remove+0x30/0x11c)
[  985.960048] [<c05a1338>] (sdio_bus_remove) from [<c049dba4>] (__device_release_driver+0x148/0x204)
[  985.969012] [<c049dba4>] (__device_release_driver) from [<c049dc80>] (device_release_driver+0x20/0x2c)
[  985.978321] [<c049dc80>] (device_release_driver) from [<c049cf3c>] (bus_remove_device+0xcc/0xf8)
[  985.987110] [<c049cf3c>] (bus_remove_device) from [<c0498fc0>] (device_del+0x13c/0x450)
[  985.995116] [<c0498fc0>] (device_del) from [<c05a1548>] (sdio_remove_func+0x1c/0x28)
[  986.002863] [<c05a1548>] (sdio_remove_func) from [<c05a0450>] (mmc_sdio_detect+0x94/0x144)
[  986.011129] [<c05a0450>] (mmc_sdio_detect) from [<c0596fc0>] (mmc_rescan+0xcc/0x430)
[  986.018875] [<c0596fc0>] (mmc_rescan) from [<c01379ec>] (process_one_work+0x1d4/0x434)
[  986.026796] [<c01379ec>] (process_one_work) from [<c0137cb0>] (worker_thread+0x64/0x5b0)
[  986.034890] [<c0137cb0>] (worker_thread) from [<c013da94>] (kthread+0x14c/0x150)
[  986.042292] [<c013da94>] (kthread) from [<c01010e8>] (ret_from_fork+0x14/0x2c)
[  986.049514] Exception stack(0xc3947fb0 to 0xc3947ff8)
[  986.054568] 7fa0:                                     00000000 00000000 00000000 00000000
[  986.062746] 7fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  986.070923] 7fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[  986.077620] ---[ end trace e1208ee6c0a8c0bf ]---
[  986.082292] wlan0: deauthenticating from 98:0d:67:15:6d:47 by local choice (Reason: 3=DEAUTH_LEAVING)
[  986.194551] mmc1: card 0001 removed
[  986.258750] sdio_read_cis: 2 callbacks suppressed
[  986.258762] mmc1: queuing unknown CIS tuple 0x01 (3 bytes)
ping: sendmsg: Network is unreachable
ping: sendmsg: Network is unreachable
ping: sendmsg: Network is unreachable

Just insmod and rmmod result:
---------------------------------

root@PI:~# insmod esp8089.ko
[   81.190402] esp8089: loading out-of-tree module taints kernel.
[   81.216053]
[   81.216053] ***** EAGLE DRIVER VER:bdf5087c3deb*****
[   81.216053]
[   81.224242] ESP8089 reset via GPIO 132
[   81.228072] sun8i-v3s-pinctrl 1c20800.pinctrl: 1c20800.pinctrl supply vcc-pe not found, using dummy regulator
[   81.364097] mmc1: card 0001 removed
[   82.532864] mmc1: queuing unknown CIS tuple 0x01 (3 bytes)
[   82.546548] mmc1: queuing unknown CIS tuple 0x1a (5 bytes)
[   82.556171] mmc1: queuing unknown CIS tuple 0x1b (8 bytes)
[   82.563025] mmc1: queuing unknown CIS tuple 0x80 (1 bytes)
[   82.569967] mmc1: queuing unknown CIS tuple 0x81 (1 bytes)
[   82.576609] mmc1: queuing unknown CIS tuple 0x82 (1 bytes)
[   82.582172] mmc1: new SDIO card at address 0001
[   82.595836] esp_sdio_dummy_probe enter
[   82.813529] esp_sdio_init power up OK
[... 34 lines cut ...]
[   93.823356] [<bf000e50>] (esp_sdio_remove [esp8089]) from [<c05a1338>] (sdio_bus_remove+0x30/0x11c)
[   93.832411] [<c05a1338>] (sdio_bus_remove) from [<c049dba4>] (__device_release_driver+0x148/0x204)
[   93.841375] [<c049dba4>] (__device_release_driver) from [<c049e9d4>] (driver_detach+0xd4/0x114)
[   93.850078] [<c049e9d4>] (driver_detach) from [<c049d19c>] (bus_remove_driver+0x4c/0xa0)
[... cut ...]
root@PI:~# [   94.484032] mmc1: card 0001 removed
[   94.540429] sdio_read_cis: 2 callbacks suppressed
[   94.540439] mmc1: queuing unknown CIS tuple 0x01 (3 bytes)
[   94.558751] mmc1: queuing unknown CIS tuple 0x1a (5 bytes)
[   94.567402] mmc1: queuing unknown CIS tuple 0x1b (8 bytes)
[   94.575796] mmc1: queuing unknown CIS tuple 0x80 (1 bytes)
[   94.581383] mmc1: queuing unknown CIS tuple 0x81 (1 bytes)
[   94.588264] mmc1: queuing unknown CIS tuple 0x82 (1 bytes)
[   94.594885] mmc1: new SDIO card at address 0001

root@PI:~#

YusufYYK, 2022-08-11:
I'm using driver with change gpio in sdio_stub.c

/* reset GPIO parameter defaults to GPIO 0 (ID_SD) on the Raspberry Pi */
static int esp_reset_gpio = 132;
module_param(esp_reset_gpio, int, 0);
MODULE_PARM_DESC(esp_reset_gpio, "ESP8089 CH_PD reset GPIO number");

al177, 2022-08-11:
800mV drop on the supply is enough that I'd throw a cap near the ESP module.

But more worrying is the breadboard. I'm confident your problem is signal integrity. Look through the closed bugs here. A fair number of them were cleared up by shortening and equalizing length of the wires. I can't speak for your SoC, but you may get better results with your setup if you can turn down the SDIO clock speed significantly.

YusufYYK, 2022-08-12:
Hi Andrew,

I focused on SI. And short connections with pullup and series resistors.i am sure about SI. After connect the board to my wifi network, i noticed that pinging from other computer to the board "No error occurs"! When i ping with ssh or serial connection inside the board to any device same error occurs. How can it be? is it kernel issue or something else? What is your opinion? Any advice?

YusufYYK, 2022-08-13:
i try download a file with big size(standby time: 1 hour). No problem occurs! 

# wget http://mirror.ufs.ac.za/linuxmint/stable/14/linuxmint-14-kde-dvd-64bit.iso
Connecting to mirror.ufs.ac.za (196.255.240.40:80)
saving to 'linuxmint-14-kde-dvd-64bit.iso'
linuxmint-14-kde-dvd 100% |********************************| 1045M  0:00:00 ETA
'linuxmint-14-kde-dvd-64bit.iso' saved

YusufYYK, 2022-08-13:
After that again ping fails. 

64 bytes from 192.168.88.150: seq=179 ttl=64 time=2.721 ms
64 bytes from 192.168.88.150: seq=180 ttl=64 time=5.508 ms
64 bytes from 192.168.88.150: seq=181 ttl=64 time=5.657 ms
64 bytes from 192.168.88.150: seq=182 ttl=64 time=5.611 ms
64 bytes from 192.168.88.150: seq=183 ttl=64 time=6.098 ms
64 bytes from 192.168.88.150: seq=184 ttl=64 time=1.914 ms
64 bytes from 192.168.88.150: seq=185 ttl=64 time=5.782 ms
[ 3259.168114] esp_sdio_remove enter
[ 3259.172651] sif_disable_irq release irq failed
[ 3259.177117] ------------[ cut here ]------------
[ 3259.181854] WARNING: CPU: 0 PID: 496 at kernel/workqueue.c:3049 __flush_work+0xac/0xb8
[ 3259.189808] Modules linked in: esp8089(O)
[ 3259.193834] CPU: 0 PID: 496 Comm: kworker/0:1 Tainted: G           O      5.4.209 #1
[ 3259.201574] Hardware name: Allwinner sun8i Family
[ 3259.206287] Workqueue: events_freezable mmc_rescan
[ 3259.211102] [<c010e510>] (unwind_backtrace) from [<c010b04c>] (show_stack+0x10/0x14)
[ 3259.218855] [<c010b04c>] (show_stack) from [<c07a4120>] (dump_stack+0x8c/0xa0)

al177, 2022-08-13:
It still looks like an SI issue. There are a bunch of things you need to check for: reflections (usually fixed w/ small value resistor on data and clock lines near the ESP), stubs (breadboard contacts make for large stubs), crosstalk, bus clock rate, and skew from clock to each data line (all wires between the Allwinner SoC and ESP must be the same length). From a hardware perspective, I'd try turning the clock rate as low as possible for the Allwinner's SDIO interface and see if that clears things up, then go from there.

`esp_sdio_remove() `is hooked to the remove callback for the driver. The SDIO driver will assume the device is dead and remove the device if it times out on a transfer that didn't complete.

You should also set up SDIO tracing before installing the module and bringing up the interface to see why it's being removed by the kernel. This is a good guide for the debug process: https://community.silabs.com/s/article/linux-sdio-detection?language=en_US

YusufYYK, 2022-08-14:
I use another software fping(open source project) then error occured but module is not removed and kernel is not crashed. It is smarter then classical ping software. It continue pinging after error. So I will buy ESP-12F and  try again. As you said it is SI problem. I thing this ESP-201 modules board SI is not good. After trying with 12F i will feedback. Thanks

192.168.88.1 : [375], 64 bytes, 1.70 ms (6.43 avg, 1% loss)
192.168.88.1 : [376], 64 bytes, 5.49 ms (6.43 avg, 1% loss)
192.168.88.1 : [377], 64 bytes, 95.4 ms (6.67 avg, 1% loss)
192.168.88.1 : [378], 64 bytes, 5.47 ms (6.67 avg, 1% loss)
192.168.88.1 : [379], 64 bytes, 5.55 ms (6.66 avg, 1% loss)
192.168.88.1 : [380], 64 bytes, 5.40 ms (6.66 avg, 1% loss)
[  419.208929] sunxi-mmc 1c10000.mmc: data error, sending stop command
[  419.215614] sunxi-mmc 1c10000.mmc: data error, sending stop command
[  419.221984] sunxi-mmc 1c10000.mmc: data error, sending stop command
[  419.228381] sunxi-mmc 1c10000.mmc: data error, sending stop command
[  419.234746] sunxi-mmc 1c10000.mmc: data error, sending stop command
[  419.241075] sunxi-mmc 1c10000.mmc: data error, sending stop command
[  419.247425] sunxi-mmc 1c10000.mmc: data error, sending stop command
[  419.304810] sunxi-mmc 1c10000.mmc: data error, sending stop command
[  419.311220] sunxi-mmc 1c10000.mmc: data error, sending stop command
[  419.317634] sunxi-mmc 1c10000.mmc: data error, sending stop command
[  419.324017] sunxi-mmc 1c10000.mmc: data error, sending stop command
[  419.330349] sunxi-mmc 1c10000.mmc: data error, sending stop command
192.168.88.1 : [381], 64 bytes, 5.40 ms (6.66 avg, 1% loss)
192.168.88.1 : [382], 64 bytes, 8.02 ms (6.66 avg, 1% loss)
192.168.88.1 : [383], 64 bytes, 7.70 ms (6.66 avg, 1% loss)
192.168.88.1 : [384], 64 bytes, 5.53 ms (6.66 avg, 1% loss)
192.168.88.1 : [385], 64 bytes, 8.19 ms (6.66 avg, 1% loss)
192.168.88.1 : [386], 64 bytes, 2.51 ms (6.65 avg, 1% loss)
192.168.88.1 : [387], 64 bytes, 5.39 ms (6.65 avg, 1% loss)

YusufYYK, 2022-08-17:
I try it with kernel 4.10.15 again same result.

YusufYYK, 2022-08-17:
I use script for be sure about ping tool. It close ping after finish and starts new process. Same result. It isn't about ping buffer or process number.

for i in $(seq 5000)
do
ping 192.168.88.1 -c 1
sleep 1
echo $i
done

YusufYYK, 2022-08-17:
It is most likely module driver issue. I think it is a bug. This bug attempt to remove driver from kernel.

al177, 2022-08-17:
I dusted off my test setup (Pi2B, ESP12F on my WiFi Pants HAT) and attempted to reproduce your results. The Pi is running the latest kernel (5.15.56) with a build from the HEAD of this project (with a minor fix to remove now-unsupported set/get_fs() calls and module filesystem access). Both ping intervals (100ms, 10ms) ran for extended periods of time without error. I also ran your ping loop for about 1000 iterations without fail. Finally I used iperf for bandwidth measurements with no issues with a 9.63MB payload with a performance of 7.8Mbits/s (AP is across the house).

During the above tests I did not see the kernel errors that you are reporting.

Without being able to reproduce this on the platform I have at hand, I'm out of suggestions for you. We can keep this open for a few months, and if you make any progress please update this issue and submit any PRs if you do find that a fix for the driver is needed.

YusufYYK, 2022-08-17:
Ok. Can you share fixed driver for kernel 5.15.56? I will test in my board again.if it fails, i will change my SoC and try with RPI or different board. I will share results after tests.

al177, 2022-08-17:
I have committed the changes and am testing a new .deb package. If you're building from source you can pull the latest.

I don't think this will change anything for you as the difference won't be ifdef'd in unless you're using 5.10.0 or later

YusufYYK, 2022-08-18:
I try with 5.15.56 and same result with ping tool. With iperf result is good. I see "sunxi-mmc 1c10000.mmc: data error, sending stop command" error but it doesn't make kernel corrupt like fping. As a result only one thing remains, is about board(licheepi) SI or SoC. I will ignore this issue. If it is about SI when i design a board it will fixed. Thank you very much for your interest Andrew.

# iperf3 -c 192.168.88.146 -t 5000
Connecting to host 192.168.88.146, port 5201
[  5] local 192.168.88.95 port 59926 connected to 192.168.88.146 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  1.68 MBytes  14.1 Mbits/sec    0    124 KBytes
[  5]   1.00-2.00   sec   954 KBytes  7.82 Mbits/sec   12    106 KBytes
[  5]   2.00-3.00   sec  1.30 MBytes  10.9 Mbits/sec    0    129 KBytes
[  5]   3.00-4.00   sec  1.30 MBytes  10.9 Mbits/sec    3    102 KBytes
..
..
[  5] 694.00-695.00 sec  1.30 MBytes  10.9 Mbits/sec    0    144 KBytes
[  5] 695.00-696.00 sec  1.12 MBytes  9.38 Mbits/sec    0    173 KBytes
[  5] 696.00-697.00 sec   817 KBytes  6.70 Mbits/sec    0    221 KBytes
[  5] 697.00-698.00 sec  1.39 MBytes  11.7 Mbits/sec    0    221 KBytes
[  5] 698.00-699.00 sec   607 KBytes  4.97 Mbits/sec   45    115 KBytes
**[ 1211.148209] sunxi-mmc 1c10000.mmc: data error, sending stop command**
[  5] 699.00-700.00 sec  1.55 MBytes  13.0 Mbits/sec    0    129 KBytes
[  5] 700.00-701.00 sec  1.12 MBytes  9.40 Mbits/sec    0    137 KBytes
[  5] 701.00-702.00 sec  1.12 MBytes  9.38 Mbits/sec    0    139 KBytes
[  5] 702.00-703.00 sec  1.30 MBytes  10.9 Mbits/sec    0    140 KBytes

- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-752.01 sec   863 MBytes  9.63 Mbits/sec  1167             sender
[  5]   0.00-752.01 sec  0.00 Bytes  0.00 bits/sec                  receiver

YusufYYK, 2022-11-07:
Hi friend,

I solve the crash problem. It is about sleep mode. When i set no_auto_sleep=1 on config of your driver the problem is gone away.

al177, 2022-11-07:
It's interesting that we don't see that behavior on the Raspberry Pi platforms. Maybe there's an edge case in this driver or the platform's SDIO driver that only shows on some platforms.

For now I'm going to close this and consider reopening if we run into this again. Glad you got it working!

closed 2022-11-07 by al177