al177/esp8089 #52: ESP8089 via SDIO - AP long start

groove-max, 2022-12-06:
Hi!
Using 5.17.15 mainline kernel with Allwinner V3s custom board.
Last month I used esp8089 as AP client connected to my home router without any problem. But yesterday I had to switch to AP node and got strange issue:
AP needs to start about 300 secs after reboot. I tried to switch between wpa_supplicant and hostapd and the problem is still persist in any case. There are no any messages or warnings in log. The only thing i noticed that I got **beacon length:%d,fc:0x%x** messages appear after ~300 secs. After this, if I run "ifdown wlan0" and  then "ifup wlan0" AP starts immediately.
Does anybody have any ideas?
```
[    0.995694] esp8089: loading out-of-tree module taints kernel.
[    0.999268]
[    0.999268] ***** EAGLE DRIVER VER:bdf5087c3deb*****
[    0.999268]
[    0.999354] ESP8089 reset via GPIO 37
[    2.377577] mmc1: queuing unknown CIS tuple 0x01 [d9 01 ff] (3 bytes)
[    2.385089] mmc1: queuing unknown CIS tuple 0x1a [01 01 00 02 07] (5 bytes)
[    2.388170] mmc1: queuing unknown CIS tuple 0x1b [c1 41 30 30 ff ff ff ff] (8 bytes)
[    2.389964] mmc1: new high speed SDIO card at address 0001
[    2.390427] esp_sdio_dummy_probe enter
[    2.608157] esp_sdio_init power up OK
[    3.066196] esp_host:bdf5087c3deb
[    3.066196] esp_target: e826c2b3c9fd 57 18202
[    3.066196]
[    3.066322] first normal exit
[    3.066480] esp_sdio_remove enter
[    3.178325] eagle_sdio: probe of mmc1:0001:1 failed with error -110
[    3.448467] mmc1: card 0001 removed
[    3.499247] mmc1: queuing unknown CIS tuple 0x01 [d9 01 ff] (3 bytes)
[    3.506250] mmc1: queuing unknown CIS tuple 0x1a [01 01 00 02 07] (5 bytes)
[    3.509419] mmc1: queuing unknown CIS tuple 0x1b [c1 41 30 30 ff ff ff ff] (8 bytes)
[    3.511030] mmc1: new high speed SDIO card at address 0001
[    3.871431] esp_host:bdf5087c3deb
[    3.871431] esp_target: e826c2b3c9fd 57 18202
[    3.871431]
[    4.448800] random: wpa_supplicant: uninitialized urandom read (4096 bytes read)
[    4.638644] esp_op_add_interface STA
[    9.308164] random: crng init done
[   10.469659] esp_op_change_interface enter,change to AP
[   10.469689] esp_op_change_interface type from 2 to 3
[   30.568174] vcc2v8: disabling
[   30.568204] vcc1v8: disabling
[   30.568210] vcc3v0: disabling
[   30.568218] vcc5v0: disabling
[  305.128172]  beacon length:250,fc:0x80
[  361.508188]  beacon length:250,fc:0x80
.....
```

al177, 2022-12-09:
Maybe something is trying to use it in STA mode at the same time you're trying to switch it to AP?

groove-max, 2022-12-11:
I did solve my problem!
When AP mode is started for the first time, uninitialized variables are used, as a result of which the timer is not set correctly.
I think it's a bug in the driver code. 
Here is the solution:
```
diff -urpN a/esp_mac80211.c b/esp_mac80211.c
--- a/esp_mac80211.c	2022-12-09 19:03:25.162009469 -0500
+++ b/esp_mac80211.c	2022-12-09 19:04:44.126865097 -0500
@@ -488,6 +488,8 @@ static void init_beacon_timer(struct iee
 	ESP_IEEE80211_DBG(ESP_DBG_OP, " %s enter: beacon interval %x\n", __func__, evif->beacon_interval);
 
 	beacon_tim_init();
+	cycle_beacon_count = 1;
+	init_jiffies = jiffies;
 #if (LINUX_VERSION_CODE >= KERNEL_VERSION(2, 6, 28))
 	evif->beacon_timer.expires = init_jiffies + msecs_to_jiffies(cycle_beacon_count * vif->bss_conf.beacon_int*1024/1000);
 #else
@@ -495,8 +497,6 @@ static void init_beacon_timer(struct iee
 #endif
 #if LINUX_VERSION_CODE < KERNEL_VERSION(4, 15, 0)
         init_timer(&evif->beacon_timer);  //TBD, not init here...
-	cycle_beacon_count = 1;
-	init_jiffies = jiffies;
 	evif->beacon_timer.data = (unsigned long) vif;
 	evif->beacon_timer.function = drv_handle_beacon;
 #else
```

al177, 2022-12-12:
Ooh. Good catch! I'll test it out and merge it in the next few days.

Thank you!

al177, 2023-03-06:
After a couple of months of distractions. applied your patch and am cutting a release now. Thanks for your contribution!

closed 2023-03-06 by al177