al177/esp8089 #54: How to troubleshoot?

ab0tj, 2023-10-27:
I am trying to integrate your driver into JELOS Linux on the Odroid Go Advance handheld system, which uses the RockChip RK3326 CPU. I managed to get the driver to build but it is still not working for me. I'm not asking you to make it work, but how do I go about troubleshooting? I get the below output but I'm not sure if this means the firmware even loaded or not. It doesn't seem to be detecting the module for the second init, though.

```[   27.956094] ***** EAGLE DRIVER VER:bdf5087c3deb*****
[   27.956094] 
[   27.956240] ESP8089 reset via GPIO 105
[   27.956279] gpio-105 (reset): gpiod_request: status -16
[   28.033717] usb 1-1: reset high-speed USB device number 2 using dwc2
[   28.160882] esp_sdio_dummy_probe enter
[   28.367994] esp_sdio_init power up OK
[   28.861871] esp_host:bdf5087c3deb
[   28.861871] esp_target: e826c2b3c9fd 57 18202
[   28.861871] 
[   28.862195] first normal exit
[   28.862542] esp_sdio_remove enter
[   28.967401] eagle_sdio: probe of mmc1:0001:1 failed with error -110
```

The wifi module works with other Linux distros (using the RockChip kernel branch, I believe), so I'm pretty sure this just has something to do with the way I'm implementing your driver.

Thanks!

al177, 2023-10-27:
It looks like the request to toggle the reset GPIO 105 is failing. I think "status -16" means that the GPIO device is busy, probably assigned to another driver or application.

ab0tj, 2023-10-27:
Ah, yeah, that. It is used by the MMC power sequencing for SDIO. Removing that took care of the error but didn't make the driver work either.

al177, 2023-10-27:
The log you posted, outside of the gpiod_request error, looks normal especially if the card firmware has already been loaded. Do you see any additional messages like "esp_op_add_interface STA" or any new interfaces like "wlan0" in the logs or "ip link"?

Have you power cycled your platform? If the driver can't reliably assert reset, then it can fail to detect that fw is already running.

ab0tj, 2023-10-27:
Unfortunately this is where the output ends from the driver. A reboot or power cycle just gives the same results. Here's the output from a different distro on the same machine that already has esp8089 support:
```
[    4.450661] 
               ***** EAGLE DRIVER VER:bdf5087c3deb*****

[    4.450746] ESP8089 reset via GPIO 105
[    4.453607] esp_sdio_dummy_probe enter
[    4.656865] esp_sdio_init power up OK
[    4.866819] r8152 1-1:1.0 eth0: v1.08.3
[    5.147489] esp_host:bdf5087c3deb
               esp_target: e826c2b3c9fd 57 18202

[    5.148044] esp_readwrite_file: file /system/lib/modules/test_results filp_open error
[    5.148154] first normal exit
[    5.148321] esp_sdio_remove enter
[    5.245326] mmc1: card 0001 removed
[    5.245724] mmc_host mmc1: Bus speed (slot 0) = 400000Hz (slot req 400000Hz, actual 400000HZ div = 0)
[    5.285907] mmc1: queuing unknown CIS tuple 0x01 (3 bytes)
[    5.293940] mmc1: queuing unknown CIS tuple 0x1a (5 bytes)
[    5.297633] mmc1: queuing unknown CIS tuple 0x1b (8 bytes)
[    5.298748] mmc_host mmc1: Bus speed (slot 0) = 50000000Hz (slot req 50000000Hz, actual 50000000HZ div = 0)
[    5.300401] mmc1: queuing unknown CIS tuple 0x80 (1 bytes)
[    5.300595] mmc1: new high speed SDIO card at address 0001
[    5.657606] esp_host:bdf5087c3deb
               esp_target: e826c2b3c9fd 57 18202

[    7.495209] NET: Registered protocol family 10
[    7.585179] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[    7.617324] esp_op_add_interface STA 
```

As far as I can tell, the biggest difference is that in the non-working setup, it never shows the mmc device being removed and then gets error -110 from eagle_sdio. This doesn't happen in the working setup. I have done tons of Google searching on the subject and my best guess at this point is that the SDIO controller isn't detecting the esp8089 being removed from the bus and thus never tries to initialize it the second time around?

al177, 2023-10-27:
That could be it. I ran into this in the past on the RPi, and there was a config.txt option for the sdio driver overlay "poll_once=off" that needed to be provided to get the SDIO interface to scan for new devices regardless of if another device was previously on the bus. I think that's specific to the Pi SDIO interface driver, so you'll have to check for a Rockchip kernel equivalent. Maybe look at /proc/cmdline, look at the devicetree sources for your platform, or look at the SDIO driver node in sysfs for any parameters

al177, 2023-10-27:
[I ran across this which may help: ](https://forum.odroid.com/viewtopic.php?t=38124)

ab0tj, 2023-10-27:
Thanks, I'll check out the link. In the meantime I did another build to free up that GPIO and this is the output:
```
[   28.100963] ***** EAGLE DRIVER VER:bdf5087c3deb*****
[   28.100963] 
[   28.101210] ESP8089 reset via GPIO 105
[   28.307564] esp_sdio_dummy_probe enter
[   28.513509] esp_sdio_init power up OK
[   29.016414] esp_host:bdf5087c3deb
[   29.016414] esp_target: e826c2b3c9fd 57 18202
[   29.016414] 
[   29.017066] first normal exit
[   29.017536] esp_sdio_remove enter
[   29.017772] sif_disable_irq release irq failed
[   29.123967] eagle_sdio: probe of mmc1:0001:1 failed with error -110
```

The main difference I see here is it now has an error about not being able to disable an irq. Otherwise looks the same to me...

al177, 2023-10-27:
There's also this patch to the integrated esp8089 driver in the Hardkernel kernel repo referencing the OdroidGoA and not rescanning after the first reset: https://github.com/hardkernel/linux/commit/3921e7a6ae36779bc05b1c1b941e545e04f7872d .

ab0tj, 2023-10-27:
Thank you!!! Adding those changes as a patch seems to have made the driver load and the wlan0 device shows up. It still isn't working (won't connect to networks), but that's progress!

ab0tj, 2023-10-27:
To anyone that comes across this in the future, I solved that last issue by adding "esp,crystal-26M-en = <2>;" to the esp8089 entry in the dtree for this device. It still does some weird things like not initializing if I have an ethernet adapter plugged in at boot, but I think it is just down to tweaking SDIO settings now.

closed 2023-10-27 by ab0tj